locka99/opcua #4: simple-server example fails due WouldBlock Error, is then stuck in loop

MattsSe, 2017-11-08:
Hi,

I tried to get started with the examples. However I had some problems with the encrypted endpoints at first. I could not connect with the opcua-commander, even if I set the securityPolicy and securityMode explicitly: `opcua-commander -e opc.tcp://localhost:4855 -P=None -s=NONE`.
There is also no error log after the simple server receives the `CreateSessionRequest`.
My guess was that this is due the exact same path (`/`) for each endpoint in the server.conf yaml.
So I removed all endpoints besides None,None from the server.conf.
Now the opcua-commander can connect and opens its gui. Securitymode and policy matches. But the session is immediately disconnect. Now the server seems to be stuck in a loop, where he receives a new connection but discards it immediately and so on.

I tracked this down to the `TcpTransport::run` function:
line: 153
```
// Try to read, using timeout as a polling mechanism
let bytes_read_result = stream.read(&mut in_buf);
```
This result is an `WouldBlock` error and yields in breaking the loop. This is the log msg:
`2017-11-08 16:52:48.232 - DEBUG - opcua_server::comms::tcp_transport - Read error - kind = WouldBlock, Error { repr: Os { code: 35, message: "Resource temporarily unavailable" } }`

The Session is now shutting down before the next session is created and discarded again...

Here is the log of one cycle:

```
2017-11-08 16:52:48.179 - INFO - opcua_server::server - Handling new connection Ok(TcpStream { addr: V4(127.0.0.1:4855), peer: V4(127.0.0.1:59838), fd: 4 })
2017-11-08 16:52:48.179 - INFO - opcua_server::comms::tcp_transport - Session started 2017-11-08 16:52:48.179447 UTC
2017-11-08 16:52:48.180 - INFO - opcua_server::server - Removing terminated session
2017-11-08 16:52:48.180 - DEBUG - opcua_server::comms::tcp_transport - Processing HELLO
2017-11-08 16:52:48.180 - DEBUG - opcua_server::comms::tcp_transport - Sending ACK
2017-11-08 16:52:48.181 - DEBUG - opcua_server::comms::tcp_transport - Processing message
2017-11-08 16:52:48.181 - DEBUG - opcua_server::comms::secure_channel_service - Message security mode == None
2017-11-08 16:52:48.232 - DEBUG - opcua_server::comms::tcp_transport - Read error - kind = WouldBlock, Error { repr: Os { code: 35, message: "Resource temporarily unavailable" } }
2017-11-08 16:52:48.232 - INFO - opcua_server::comms::tcp_transport - Session terminating normally, session_status_code = GOOD
2017-11-08 16:52:48.232 - INFO - opcua_server::comms::tcp_transport - Terminating socket
2017-11-08 16:52:48.232 - INFO - opcua_server::comms::tcp_transport - Session is finished Duration { secs: 0, nanos: 53278000 }
2017-11-08 16:52:48.232 - INFO - opcua_server::server - Session thread is terminated
```

Any Ideas?

locka99, 2017-11-09:
I'll take a look tonight and see if I can identify the error. I've been using Softing's free OPC UA client quite successfully and as well as the .NET and Eclipse open stack clients. They all tend to reveal different things depending on how stringently they check the connection. I've only used NodeJS with no crypto but I'll see if I can get opc commander going on crypto.

I have previously seen the server get into a loop with a high CPU from a malformed connection but was unable to identity the cause.

As far as I know it's okay to create different endpoints with the same path. The list of endpoints is informational so it doesn't matter if they contain duplicates. What matters is that the client connects with OpenSecureChannel (security policy & message mode), CreateSession (path) and ActivateSession (user token) that it matches an endpoint.

MattsSe, 2017-11-09:
Yes, multiple endpoints should not matter.
I'm on mac and I also use the (free opcua-client)[https://github.com/FreeOpcUa/opcua-client-gui]. I could establish an encrypted connection via a certificate successfully after trusting the file, however the created session is closed again instantly due to the same WouldBlock error in tcp_transport.

```
2017-11-09 14:42:23.961 - DEBUG - opcua_server::server_state - find_endpoint, url = opc.tcp://localhost:4855
2017-11-09 14:42:23.964 - DEBUG - opcua_core::crypto::certificate_store - Validating cert with name on disk OPC UA Demo Key [ed78ec35cf2ad1b74d49dc2074e2420a6e265f14].der
2017-11-09 14:42:23.968 - DEBUG - opcua_server::comms::tcp_transport - Processing message
2017-11-09 14:42:23.968 - DEBUG - opcua_core::comms::secure_channel - Decrypting block with signature info 0..444 and encrypt info 16..464
2017-11-09 14:42:23.970 - DEBUG - opcua_server::server_state - Anonymous identity is authenticated
2017-11-09 14:42:23.974 - DEBUG - opcua_server::comms::tcp_transport - Processing message
2017-11-09 14:42:23.974 - DEBUG - opcua_core::comms::secure_channel - Decrypting block with signature info 0..174 and encrypt info 16..194
2017-11-09 14:42:23.987 - DEBUG - opcua_server::comms::tcp_transport - Processing message
2017-11-09 14:42:23.987 - DEBUG - opcua_core::comms::secure_channel - Decrypting block with signature info 0..133 and encrypt info 16..153
2017-11-09 14:42:24.037 - DEBUG - opcua_server::comms::tcp_transport - Read error - kind = WouldBlock, Error { repr: Os { code: 35, message: "Resource temporarily unavailable" } }
2017-11-09 14:42:24.037 - INFO - opcua_server::comms::tcp_transport - Session terminating normally, session_status_code = GOOD
2017-11-09 14:42:24.037 - INFO - opcua_server::comms::tcp_transport - Terminating socket
2017-11-09 14:42:24.037 - INFO - opcua_server::comms::tcp_transport - Session is finished Duration { secs: 0, nanos: 94825000 }
2017-11-09 14:42:24.037 - INFO - opcua_server::server - Session thread is terminated
```

locka99, 2017-11-09:
I can reproduce in opcua-commander so I'll investigate. Similar to what I've seen in the past - high CPU on the session thread and the client drops out.

locka99, 2017-11-10:
I've changed the code to loop on WouldBlock. 

The problem with opcua-commander and endpoints is still there. I had to comment out a line in opcua-commander/index.js that invokes exit(-1), to see what it was printing out when DEBUG=ALL env is set. At that point it appears to be that opcua-commander is looking for an endpoint that supports an anonymous useridentity token and not finding it. My endpoint descriptions do include this token and a policy id so I have yet to figure if the problem is on my side, e.g. with ordering, or opcua-commander side. 

The workaround for now is to edit samples/sample.conf and just remove all endpoints except the one with the id "none". Then you should be able to connect.

locka99, 2017-11-12:
Okay I understand the remaining issue and have a simple resolution to it

The command line:

```opcua-commander -e opc.tcp://localhost:4855 -P=None -s=NONE```

But it should actually be :

```opcua-commander -e opc.tcp://127.0.0.1:4855/ -P=None -s=NONE```

The sample server config sets the host to be 127.0.0.1 so all the endpoints are opc.tcp://127.0.0.1:4855/ etc. In addition the sample config has an endpoint called /noaccess which is simply there to test authentication behaviour.

So what is happening is:

1. opcua-commander (node-opcua) calls GetEndpoints and grabs a list of endpoints from my server, all of which have 127.0.0.1 in them
2. It tries to find a matching url for opc.tcp://localhost:4855/ with None/None security but that fails. (https://github.com/node-opcua/node-opcua/blob/36fa3653ff1132128a36ccc59d41cf4eded648f7/packages/node-opcua-client/src/opcua_client.js#L124)
3. Then fallback strategy is to find the first url on the server which has None/None security (https://github.com/node-opcua/node-opcua/blob/36fa3653ff1132128a36ccc59d41cf4eded648f7/packages/node-opcua-client/src/opcua_client.js#L138)
4. Unfortunately that happens to be /noaccess which has no anonymous or userpass id
5. opcua-commander then blows up in the activate session when it discovers the endpoint has no anonymous token.

So the resolution is to either:

1. Use the IP address from the command line
2. Edit the server.conf and put localhost (or any other resolvable name) into the tcp_config/host field.

node-opcua should really ignore the hostname when doing matches as a fallback mechanism rather than grabbing the first entry point with a matching policy.

closed 2017-11-12 by locka99