locka99/opcua #9: Add serde serialization

denizs, 2018-06-02:
Hey there,
are you planning on adding serde serialization support any time soon? Right now, I'm following the path described [here](https://serde.rs/remote-derive.html), but this is really tedious, as there are multiple layers of composed structs 👎

locka99, 2018-06-02:
I have serde serialisation of configuration files. What other serialisation were you wanting?

denizs, 2018-06-02:
I need to serialize `DataValue`s to send them to a server via MQTT. Right now, I'm using serde's `remote` and `with` macro feature, but getting this out of the box would be awesome. If you like, I could start working on a PR :)

locka99, 2018-06-02:
It shouldn't be hard to add serialize / deserialize derivations aove the DataValue type. If you want to go a step further you could even implement the samples/mqtt-client which I added as a placeholder last week but hadn't gotten around to writing :)

locka99, 2018-06-03:
I've checked in Serialize / Deserialize impls for Variant, DataValue and everything a Variant can hold. The only tricky types were Guid and DateTime which needed some manual impls.

I hope this is enough. If you are doing MQTT it would be nice to have a sample publish client to hook up in my samples

denizs, 2018-06-04:
Ok awesome 👍  I'll start working on it today!

denizs, 2018-06-18:
Hey there, sorry for the slight delay. I just waned to start working on the serialization and deserializaion functionality, but it seems like you already included this in `master` 🎉.  Thanks a lot! I will start implementing an MQTT example later today.

locka99, 2018-06-18:
I'll open a bug about mqtt. There is a mqtt-client sample there with a couple of TODO points where publish would probably go.

closed 2018-06-18 by locka99