locka99/opcua #12: error: failed to run custom build command for `ring v0.12.1`

dergroncki, 2018-09-23:
Operating System: Windows 10 Pro

Rust toolchain:
nightly-x86_64-pc-windows-msvc (default)
rustc 1.30.0-nightly (63c75d375 2018-09-21)

I am trying to compile and run the opc ua client but I am getting the following error:

`   Compiling ring v0.12.1
error: failed to run custom build command for `ring v0.12.1`
process didn't exit successfully: `C:\Users\micha\Documents\Rust\Libraries\opcua\target\debug\build\ring-d150b23a3701555a\build-script-build` (exit code: 101)
--- stderr
thread '<unnamed>' panicked at 'execution failed', C:\Users\micha\.cargo\registry\src\github.com-1ecc6299db9ec823\ring-0.12.1\build.rs:636:9
thread '<unnamed>' panicked at 'execution failed', C:\Users\micha\.cargo\registry\src\github.com-1ecc6299db9ec823\ring-0.12.1\build.rs:636:9`

Regarding this [issue](https://github.com/briansmith/ring/issues/607) the problem should be solved with ring-0.13.

I would appreciate any help.

locka99, 2018-09-23:
I've updated the head to use ring 0.13. I don't have MSVC set up to test because there are other issues with OpenSSL that are easier to avoid by using MingW, but let me know if it helps.

dergroncki, 2018-09-23:
Thanks. Compiling of simple-client and simple-server is working now.

closed 2018-09-23 by dergroncki