locka99/opcua #18: Issue when using endpoints with private hostnames

axelfaure, 2018-12-20:
Hello !

I found what I think is a bug in the way this library handles endpoints (but maybe I am wrong and there is a way to bypass this behavior).

I try to connect an OPC-UA server using a public hostname, but this server exposes endpoints that match private hostnames on its network. 
Unfortunately, the library attempts to resolve the private hostname and since it cannot it fails to use it.

```
[opcua-client-0.5.0/src/client.rs:opcua_client::client] [ERROR] Got an error while creating the default session - The Server does not recognize the QueryString specified. [ERROR] Invalid address privatehost.mycompany.com:49320, cannot be parsed Custom { kind: Other, error: StringError("failed to lookup address information: Name or service not known") }
[opcua-client-0.5.0/src/client.rs:opcua_client::client] [ERROR] Got an error while creating the default session - The Server does not recognize the QueryString specified.
```

I managed to "solve" the issue by setting the hostname in my /etc/hosts and associating it with the server public IP but I think the library should always do it without trying to resolve the hostname (other clients and libraries seem to do that at least).

Please note that I am not at all an expert in OPC-UA and I may have misunderstood how things are expected to work.

Thank you !

locka99, 2018-12-20:
Issue #17 is similar to this too. I'm playing around with the client api at the moment to make it a little simpler to manage.

The standard way to connect to the OPC UA server is.

1. Define the endpoint(s) you want to connect to in your client config, including id, security, identity token.
2. Connect by the id
3. Internally the code will fetch endpoints from the server
4. Internally the code will match your desired endpoint to those available and connect to it.

But it does not work if the hostname does not match. I may tweak the config / API to allow fuzzy matching, i.e. connect even if the hostname doesn't work.

I've also changed the API on master to support a more ad hoc connection where you don't have to define the endpoint you wish to connect to - you just connect via a url and security policy. This only works with anonymous connections at present but if you're using the master branch it might prove useful. The simple-client demo uses this method now:

https://github.com/locka99/opcua/blob/master/samples/simple-client/src/main.rs#L44

axelfaure, 2018-12-21:
Thank you for the explanations and help!

The ad-hoc connection sounds like a good solution in our case (that's what we were doing previously using a python library).

I would be happy to give you a more detailed feedbacks on what is currently on master branch but unfortunately the only server I can access at the moment requires authenticated connections.

axelfaure, 2019-01-03:
Sorry about the delay, I took some days off.

I tried something and got a panic in return. However, I am not totally sure I configured everything as expected.
This is a very quick test, I will try to investigate it a bit more when I have some time.

In the meantime, here is a simplified version of my code:

```rust
let mut client = ClientBuilder::new()
    .application_name("Simple Client")
    .application_uri("urn:SimpleClient")
    .client().unwrap();

// I am not sure what user token policy should be, anonymous is probably a mistake
let endpoint_tuple = (url.as_ref(), SecurityPolicy::Basic256.to_str(), MessageSecurityMode::SignAndEncrypt, UserTokenPolicy::anonymous());
let id_token = IdentityToken::UserName( "<user>".to_string(), "<pass>".to_string() );
if let Ok( session ) = client.connect_to_endpoint(endpoint_tuple, id_token) {

    let node_list :Vec<ReadValueId> = vec![
        ReadValueId::from( NodeId::new( 2, "s=<path>.<entity>.<entry>" ) )
    ];

    let mut session_locked = session.write().unwrap();
    
    // Panic happens when executing this line
    let res = session_locked.read_nodes( &node_list );

}
```

The panic message I get is the following:

```
thread 'tests::it_works' panicked at 'called `Result::unwrap()` on an `Err` value: "PoisonError { inner: .. }"', libcore/result.rs:1009:5
[session.rs:opcua_client::session] [INFO] Session has dropped
thread 'tests::it_works' panicked at 'called `Result::unwrap()` on an `Err` value: "PoisonError { inner: .. }"', libcore/result.rs:1009:5
stack backtrace:
   0:     0x561a1f7fb51f - std::sys::unix::backtrace::tracing::imp::unwind_backtrace::hab66bb8ff7088f29
                               at libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
   1:     0x561a1f805987 - std::sys_common::backtrace::print::h56719ed580ab1bba
                               at libstd/sys_common/backtrace.rs:71
                               at libstd/sys_common/backtrace.rs:59
   2:     0x561a1f80023f - std::panicking::default_hook::{{closure}}::hf8d811a2c97e4b3a
                               at libstd/panicking.rs:211
   3:     0x561a1f7fffa4 - std::panicking::default_hook::hc6e051251a5e6b4e
                               at libstd/panicking.rs:227
   4:     0x561a1f80091e - std::panicking::rust_panic_with_hook::h71214e7ce0f7ac01
                               at libstd/panicking.rs:476
   5:     0x561a1f8004c1 - std::panicking::continue_panic_fmt::ha8b8442f4ea9bcac
                               at libstd/panicking.rs:390
   6:     0x561a1f8003a5 - rust_begin_unwind
                               at libstd/panicking.rs:325
   7:     0x561a1f84817c - core::panicking::panic_fmt::h0c93626b89c38af6
                               at libcore/panicking.rs:77
   8:     0x561a1eaf892d - core::result::unwrap_failed::ha3dab546f4bb9281
                               at libcore/macros.rs:26
   9:     0x561a1eaf5484 - <core::result::Result<T, E>>::unwrap::h74eb18b566b5c504
                               at libcore/result.rs:808
  10:     0x561a1ead618b - opcua_client::session::Session::make_request_header::h42a3c9ad33b07bd2
                               at /home/user/.cargo/git/checkouts/opcua-49af4c4898b3317f/9b511d6/client/src/session.rs:1113
  11:     0x561a1eacff7e - opcua_client::session::Session::close_secure_channel::hfc5836bb8e58730d
                               at /home/user/.cargo/git/checkouts/opcua-49af4c4898b3317f/9b511d6/client/src/session.rs:296
  12:     0x561a1eacfcde - opcua_client::session::Session::disconnect::hf5c4434a0f786382
                               at /home/user/.cargo/git/checkouts/opcua-49af4c4898b3317f/9b511d6/client/src/session.rs:273
  13:     0x561a1eace2b1 - <opcua_client::session::Session as core::ops::drop::Drop>::drop::hc26fa1a754bbe7c9
                               at /home/user/.cargo/git/checkouts/opcua-49af4c4898b3317f/9b511d6/client/src/session.rs:108
  14:     0x561a1eb12f94 - core::ptr::drop_in_place::hc15a84da314716d3
                               at libcore/ptr.rs:194
  15:     0x561a1eb10c6d - core::ptr::drop_in_place::h8c3d9d37c765fbde
                               at libcore/ptr.rs:194
[... 31 lines cut ...]
                               at libstd/panicking.rs:289
                               at libstd/panic.rs:392
                               at libtest/lib.rs:1423
                               at libstd/sys_common/backtrace.rs:136
  31:     0x561a1eaaf9a4 - std::panicking::try::do_call::h41920762ece46401
                               at libstd/thread/mod.rs:409
                               at libstd/panic.rs:313
                               at libstd/panicking.rs:310
  32:     0x561a1f810bb9 - __rust_maybe_catch_panic
                               at libpanic_unwind/lib.rs:102
  33:     0x561a1ea9cbcc - <F as alloc::boxed::FnBox<A>>::call_box::h36495f2c1a4be2b6
                               at libstd/panicking.rs:289
                               at libstd/panic.rs:392
                               at libstd/thread/mod.rs:408
                               at liballoc/boxed.rs:672
  34:     0x561a1f7fdeed - std::sys_common::thread::start_thread::h5213f803a61d7811
                               at liballoc/boxed.rs:682
                               at libstd/sys_common/thread.rs:24
  35:     0x561a1f7edd75 - std::sys::unix::thread::Thread::new::thread_start::he89121f566d2a8c7
                               at libstd/sys/unix/thread.rs:90
  36:     0x7ff16274bf29 - start_thread
  37:     0x7ff162465ede - __clone
  38:                0x0 - <unknown>
thread panicked while panicking. aborting.
```

axelfaure, 2019-01-18:
We finally found time to do a second test. A coworker of mine fixed my earlier test and managed to get rid of the panic. 

As you expected, the X509 Certificate now gets rejected because of the hostname.

Here are some logs of the issue

```
2019-01-17T16:40:39.506703107+01:00 [main] [opcua_client::client] [INFO] Server has these endpoints:
  opc.tcp://<endpoint_address_1>:49320 - Basic256 / SignAndEncrypt
  opc.tcp://<endpoint_address_2>:49321 - None / None
2019-01-17T16:40:39.506976937+01:00 [main] [opcua_client::client] [INFO] Creating a session for endpoint opc.tcp://<server_address>:49320/, Basic256 / SignAndEncrypt
2019-01-17T16:40:39.507382655+01:00 [main] [opcua_client::session] [INFO] Connect
2019-01-17T16:40:39.507462198+01:00 [main] [opcua_client::session] [INFO] Security policy = Basic256
2019-01-17T16:40:39.507494014+01:00 [main] [opcua_client::session] [INFO] Security mode = SignAndEncrypt
2019-01-17T16:40:39.607787722+01:00 [main] [opcua_client::session_state] [INFO] Making secure channel request
2019-01-17T16:40:39.607866418+01:00 [main] [opcua_client::session_state] [INFO] security_mode = SignAndEncrypt
2019-01-17T16:40:39.607897313+01:00 [main] [opcua_client::session_state] [INFO] security_policy = Basic256
2019-01-17T16:40:39.708724083+01:00 [main] [opcua_core::crypto::x509] [INFO] Certificate is valid for this time
2019-01-17T16:40:39.708785566+01:00 [main] [opcua_core::crypto::x509] [ERROR] Cannot find a matching hostname for input <server_address>
2019-01-17T16:40:39.935893030+01:00 [main] [opcua_core::crypto::certificate_store] [INFO] Writing X509 cert to <...>/pki/rejected/<cert_path>.der
2019-01-17T16:40:39.936014820+01:00 [main] [opcua_client::session] [ERROR] Server's certificate was rejected
2019-01-17T16:40:40.007195318+01:00 [main] [opcua_client::client] [ERROR] Got an error while creating the default session - The HostName used to connect to a Server does not match a HostName in the Certificate.

```

axelfaure, 2019-03-01:
Hi ! 

I managed to patch the library to make it work.
 Basically, the only thing needed is to extract `hostname` from the `application_uri` instead of the `endpoint_url` in `client::session::Session::create_session`

However, I only have access to a single server so I am not sure whether it is a generic solution or if it fixes my specific use case but breaks the library for other servers. Maybe you will have a better understanding of the potential impact of this change.

axelfaure, 2020-06-08:
For the record, the issue mentioned here is still relevant. I encountered it with another server (and I think it happens when the server is behind a NAT).

My fix is available here https://github.com/axelfaure/opcua/commit/47b7b0de4581133fd4360a53c00ad9172765857e 

I could create a pull request but I am really not sure if this is a sound modification or if it has other consequences (I still haven't found the time to dive into OPC-UA internal mechanisms)

locka99, 2020-06-10:
Hi, sorry for not looking at this sooner, but it looks like you're trying to infer the hostname from the application uri but the OPC UA spec describes the application uri as a globally unique identifier which could be a hostname but could equally but a guid or something else.

The original code was using the end point url and extracting the hostname - as it is advertised by the server from getEndpoints - from within that url by parsing it out. e.g. if the endpoint url says "opc.tcp://foo:1234/some/path" then hostname_from_url() should pull "foo" out from it. If "foo" is not in the cert then it's going to reject the cert.

Are you able to connect to the server from other OPC UA clients with the same security settings successfully? I'm not ruling out that the code is misinterpreting something here and it may well be but it would be useful to narrow down what works and what doesn't.

One thing you could try is to fork the code, and edit the line in session.rs that calls validate_or_reject_application_instance_cert() and pass None instead of Some(hostname). If you do that then it shouldn't look at the hostname at all when validating the cert.

axelfaure, 2020-06-11:
Hi !
Yes, other clients do connect correctly through these NATs using the exact same settings. (I even think the python library freeopcua worked in this case but I am not 100% sure though, I will try to confirm it)

Since the two servers are different and maintained be different organizations I am pretty sure that using endpoint to get hostname is not the way to go (I don't know if it is part of the spec or just the way real life servers are often dealing with NATs though). I guess providing a way to disable hostname verification altogether would work (I will test your suggestion as soon as I can find a slot for that). My guess would be that there is probably a more secure way to deal with this situation but I don't really know.

From a network point of view, I captured network traffic with wireshark and I can confirm that if you try to contact server `a.b.c.d:42` through nat server `w.x.y.z:84` and you list endpoints, the response message you receive does contain something like `opc.tcp://w.x.y.z:84/MyEndPoint` which will most likely not match the certificate of the server. (That happens also with the other clients that do work, so the issue is really in the way the certificate is validated, not something we do before that point).

As the side note, with one server my "fix" (although I understand it is probably not a good one) does work and I can connect and use the server. But for the other one, the server rejects my message when I try to activate the session (the server I try to reach logs "BadSecurityChecksFailed : Unexpected signature algorithm"). 
This happens after the server verification on client side so I think it is a completely separate issue but that is probably worth mentioning to give a complete picture.
(I will probably open a ticket for that but I am afraid I won't be able to provide much information at that point except for the fact that there is probably an issue somewhere)