locka99/opcua #19: `BadSessionIdInvalid` error on subscription

denizs, 2019-01-05:
Hey there, 
I believe this is somewhat more of a documentation topic, but I just noticed, that my subscription / session breaks, as soon as I choose a sufficiently high `publishing_interval` when creating a subscription. Working with `1f64` works fine, while choosing e.g. `1000f64` invalidates the session. 

```rust
// changing the first arg to 1000f64 invalidates the code
let subscription_id = session.create_subscription(1f64, 10, 30, 0, 0, true, callback)?
```

As the method also takes in a `lifetime_count` and `max_keep_alive_count` argument, I was wondering whether I can solve this issue by setting these properly. In case this does the job, I'd post a PR for a more elaborate documentation of `create_subcription`.
Best,
D

locka99, 2019-01-06:
Some of the client side API assumes familiarity with OPC UA and the function documentation is pretty weak. I'll add some extra comments from the specification which you can see for yourself here:

https://opcfoundation.org/developer-tools/specifications-unified-architecture/part-4-services/

Or if you prefer, here is a blog that describes the args. 

https://www.prosysopc.com/blog/opc-ua-sessions-subscriptions-and-timeouts/

I'm not sure why the server would reject a subscription with 1000 ms. It is entirely possible you've uncovered a bug, but even my sample code uses that value to connect to an NodeJS OPC UA Server with no issues. 

Basically under the covers, a timer runs every 1000ms, sending a PublishRequest to the server. The server will fill a PublishResponse with any changes to the subscription and in turn your callback is called. If no changes have happened then no response is returned until either a) a change occurs or b) keep-alive count happens. So if max keep alive is 30 ensures that at least every 30*1000ms you will get an empty PublishResponse if.

So I suppose it's possible that if the server has an inactivity timeout of less than 30 seconds, and your subscription is not changing much that the session will timeout. Therefore, you could try lowering the max keep alive count to 5 or 10 and see if it fixes the issue.

denizs, 2019-01-08:
Perfect! Thanks a lot 🎉 
I'll try this within the next few days and let you know, whether this helped us solve this issue :)

denizs, 2019-01-14:
Just found the time to experiment a little - even when I turn up the `publishing_interval` to `1000.` and the `max_keep_alive` down to `1`, I get the following output with logs in debug:

```shell
s10      | 2019-01-14 17:53:37.285 - DEBUG - opcua_client::session - Subscription timer for 3921763664 is sending a publish
s10      | 2019-01-14 17:53:37.285 - DEBUG - opcua_client::session_state - async_publish with 1 subscription acknowledgements
s10      | 2019-01-14 17:53:37.285 - DEBUG - opcua_client::session_state - async_publish, request sent with handle 10
s10      | 2019-01-14 17:53:37.285 - DEBUG - opcua_client::message_queue - Request 10 was processed by the server
s10      | 2019-01-14 17:53:37.328 - DEBUG - opcua_client::session - Processing 1 async messages
s10      | 2019-01-14 17:53:37.328 - DEBUG - opcua_client::session - handle_publish_response
s10      | 2019-01-14 17:53:37.328 - DEBUG - opcua_client::session - PublishResponse
s10      | 2019-01-14 17:53:38.243 - DEBUG - opcua_client::message_queue - Response to Request 10 has been stored
s10      | 2019-01-14 17:53:38.285 - DEBUG - opcua_client::session - Subscription timer for 3921763664 is sending a publish
s10      | 2019-01-14 17:53:38.286 - DEBUG - opcua_client::session_state - async_publish with 1 subscription acknowledgements
s10      | 2019-01-14 17:53:38.286 - DEBUG - opcua_client::session_state - async_publish, request sent with handle 11
s10      | 2019-01-14 17:53:38.286 - DEBUG - opcua_client::message_queue - Request 11 was processed by the server
s10      | 2019-01-14 17:53:38.299 - DEBUG - opcua_client::session - Processing 1 async messages
s10      | 2019-01-14 17:53:38.299 - DEBUG - opcua_client::session - handle_publish_response
s10      | 2019-01-14 17:53:38.299 - DEBUG - opcua_client::session - PublishResponse
s10      | 2019-01-14 17:53:39.244 - DEBUG - opcua_client::message_queue - Response to Request 11 has been stored
s10      | 2019-01-14 17:53:39.285 - DEBUG - opcua_client::session - Subscription timer for 3921763664 is sending a publish
s10      | 2019-01-14 17:53:39.285 - DEBUG - opcua_client::session_state - async_publish with 1 subscription acknowledgements
s10      | 2019-01-14 17:53:39.285 - DEBUG - opcua_client::session_state - async_publish, request sent with handle 12
s10      | 2019-01-14 17:53:39.285 - DEBUG - opcua_client::message_queue - Request 12 was processed by the server
s10      | 2019-01-14 17:53:39.331 - DEBUG - opcua_client::session - Processing 1 async messages
s10      | 2019-01-14 17:53:39.331 - DEBUG - opcua_client::session - handle_publish_response
s10      | 2019-01-14 17:53:39.331 - DEBUG - opcua_client::session - PublishResponse
s10      | 2019-01-14 17:53:40.240 - DEBUG - opcua_client::message_queue - Response to Request 12 has been stored
s10      | 2019-01-14 17:53:40.263 - DEBUG - opcua_client::session - Processing 1 async messages
s10      | 2019-01-14 17:53:40.263 - DEBUG - opcua_client::session - handle_publish_response
s10      | 2019-01-14 17:53:40.263 - DEBUG - opcua_client::session - PublishResponse
s10      | 2019-01-14 17:53:40.300 - DEBUG - opcua_client::session - Subscription timer for 3921763664 is sending a publish
s10      | 2019-01-14 17:53:40.300 - DEBUG - opcua_client::session_state - async_publish with 2 subscription acknowledgements
s10      | 2019-01-14 17:53:40.300 - DEBUG - opcua_client::session_state - async_publish, request sent with handle 13
s10      | 2019-01-14 17:53:40.301 - DEBUG - opcua_client::message_queue - Request 13 was processed by the server
s10      | 2019-01-14 17:53:40.303 - DEBUG - opcua_client::message_queue - Response to Request 13 has been stored
s10      | 2019-01-14 17:53:40.401 - DEBUG - opcua_client::session - Processing 1 async messages
s10      | 2019-01-14 17:53:40.401 - DEBUG - opcua_client::session - handle_publish_response
s10      | 2019-01-14 17:53:40.401 - DEBUG - opcua_client::session - Service fault received with BadSessionIdInvalid error code
s10      | 2019-01-14 17:53:41.044 - DEBUG - rumqtt::s10nection - *** PubAck --> Pkid(2)
s10      | --- Publish Queue =
s10      | [
s10      |     topic = TopicName("/events"), qos = Level1(2), retain = false, payload size = 6275 bytes
s10      | ]
s10      |
s10      |
s10      | 2019-01-14 17:53:41.044 - DEBUG - rumqtt::s10nection - Pub Q Len After Ack @@@ 0
s10      | 2019-01-14 17:53:41.285 - DEBUG - opcua_client::session - Subscription timer for 3921763664 is sending a publish
s10      | 2019-01-14 17:53:41.285 - DEBUG - opcua_client::session_state - async_publish with 0 subscription acknowledgements
s10      | 2019-01-14 17:53:41.285 - DEBUG - opcua_client::session_state - async_publish, request sent with handle 14
s10      | 2019-01-14 17:53:41.285 - DEBUG - opcua_client::message_queue - Request 14 was processed by the server
s10      | 2019-01-14 17:53:41.286 - DEBUG - opcua_client::message_queue - Response to Request 14 has been stored
s10      | 2019-01-14 17:53:41.303 - DEBUG - opcua_client::session - Processing 1 async messages
s10      | 2019-01-14 17:53:41.303 - DEBUG - opcua_client::session - handle_publish_response
s10      | 2019-01-14 17:53:41.303 - DEBUG - opcua_client::session - Service fault received with BadSessionIdInvalid error code
s10      | 2019-01-14 17:53:42.285 - DEBUG - opcua_client::session - Subscription timer for 3921763664 is sending a publish
[... cut ...]
  session_name: UAString { value: Some("Rust OPCUA Client") },
  client_nonce: ByteString { value: None },
  client_certificate: ByteString { value: Some([my-cert]) },
  requested_session_timeout: 0.0,
  max_response_message_size: 0
}
```

Any ideas what's happening there? I see that the session timeout is set to 0.0. I have two threads, one publisher and the opc thread, which are communicating via a `SyncSender`. Could it be that while waiting for the data to be sent from the opc to the publisher thread, the session simply times out?

closed 2019-02-13 by denizs