locka99/opcua #22: (auto) reconnect when server goes down

denizs, 2019-01-25:
I'm currently testing edge cases against a small python server setup. When killing the server process and restarting it, the client does not seem to be able to reconnect. That said, I'm not quite sure, whether it notices that the server has been disconnected. Is there anyway to catch this? (I tried the connection state changed and session closed callbacks, but they're not being called though).

**Log output:**

```shell
2019-01-25 10:54:40.071 - INFO - opcua_client::comms::tcp_transport - ReadState has dropped
2019-01-25 10:54:40.071 - INFO - opcua_client::comms::tcp_transport - Read loop finished
2019-01-25 10:54:41.581 - ERROR - opcua_client::comms::tcp_transport - Write IO error Os { code: 32, kind: BrokenPipe, message: "Broken pipe" }
2019-01-25 10:54:41.582 - ERROR - opcua_client::comms::tcp_transport - Write loop is finished with an error ()
2019-01-25 10:54:41.582 - INFO - opcua_client::comms::tcp_transport - WriteState has dropped
2019-01-25 10:55:09.579 - INFO - opcua_client::session_state - Making secure channel request
2019-01-25 10:55:09.579 - INFO - opcua_client::session_state - security_mode = None
2019-01-25 10:55:09.579 - INFO - opcua_client::session_state - security_policy = None
2019-01-25 10:55:19.608 - INFO - opcua_client::session_state - Timeout waiting for response from server
2019-01-25 10:55:19.609 - INFO - opcua_client::message_queue - Request 53 has timed out and any response will be ignored
2019-01-25 10:55:19.609 - INFO - opcua_client::session_state - Making secure channel request
2019-01-25 10:55:19.609 - INFO - opcua_client::session_state - security_mode = None
2019-01-25 10:55:19.609 - INFO - opcua_client::session_state - security_policy = None
```

The making secure channel request with sec mode and policy endlessly loop after that point.

locka99, 2019-01-26:
I put some session disconnect / retry / reconnect code into the crate but it doesn't work fully yet.

locka99, 2019-01-29:
I'll take a look at this tonight and see if I can reproduce. If it's just a case of setting a flag in the map_err() then it should be relatively simple to bring down the other tasks by setting the connection state to finished.

e.g.

set_connection_state!(connection_state, ConnectionState::Finished(BadConnectionClosed);

This probably has to be done in the reader and writer, but only if the connection state is not already in a finished state.

When the connection state is set to Finished, the Session::is_connected() returns false, which should mean the reconnect_and_activate() code gets called. This code hasn't been finished so it will probably reveal other errors that need to be fixed.

locka99, 2019-01-29:
BTW, the idea is there will be a SessionRetryPolicy associated with the session which says what to do on disconnect. The policy will say how many times to retry connecting and at what interval.

denizs, 2019-01-30:
So basically, the error that is revealed is, that if the server does not restore state, the sequential numbers of the `secure_channel_id` are out of sync and the secure channel is not renewed. If you can point me to the right direction, I'll be glad to help you with this, as of right now, the connection state is properly being propagated through to the session.

That said, I have used the `BadServerHalted` code. I'll try this with `BadConnectionClosed` and will have a look what happens.

denizs, 2019-01-30:
I just replaces the `ConnectionState::BadServerHalted` by  `ConnectedState::BadConnectionClosed` and got this log output:

```bash
2019-01-30 10:04:11.203 - INFO - opcua_client::session_state - Timeout waiting for response from server
2019-01-30 10:04:11.204 - INFO - opcua_client::message_queue - Request 52 has timed out and any response will be ignored
2019-01-30 10:04:11.204 - DEBUG - opcua_client::session_state - async_publish, request sent with handle 51
2019-01-30 10:04:11.204 - DEBUG - opcua_client::session - Subscription timer for 78 is sending a publish
2019-01-30 10:04:11.204 - DEBUG - opcua_client::session_state - async_publish with 0 subscription acknowledgements
2019-01-30 10:04:11.204 - INFO - opcua_client::session_state - Making secure channel request
2019-01-30 10:04:11.204 - DEBUG - opcua_client::comms::tcp_transport - Waiting for a connect (or failure to connect)
2019-01-30 10:04:11.204 - INFO - opcua_client::session_state - security_mode = None
2019-01-30 10:04:11.204 - DEBUG - opcua_client::comms::tcp_transport - Client tokio tasks are starting for connection
2019-01-30 10:04:11.204 - INFO - opcua_client::session_state - security_policy = None
2019-01-30 10:04:11.206 - DEBUG - opcua_client::comms::tcp_transport - Sending HELLO
2019-01-30 10:04:21.253 - INFO - opcua_client::session_state - Timeout waiting for response from server
2019-01-30 10:04:21.253 - INFO - opcua_client::message_queue - Request 54 has timed out and any response will be ignored
2019-01-30 10:04:21.253 - DEBUG - opcua_client::session_state - async_publish, request sent with handle 53
2019-01-30 10:04:21.253 - DEBUG - opcua_client::session - Subscription timer for 78 is sending a publish
2019-01-30 10:04:21.253 - DEBUG - opcua_client::session_state - async_publish with 0 subscription acknowledgements
2019-01-30 10:04:21.253 - INFO - opcua_client::session_state - Making secure channel request
2019-01-30 10:04:21.253 - INFO - opcua_client::session_state - security_mode = None
2019-01-30 10:04:21.253 - INFO - opcua_client::session_state - security_policy = None
2019-01-30 10:04:21.255 - DEBUG - opcua_client::comms::tcp_transport - Got ack AcknowledgeMessage { message_header: MessageHeader { message_type: Acknowledge, message_size: 28 }, protocol_version: 0, receive_buffer_size: 65536, send_buffer_size: 65536, max_message_size: 0, max_chunk_count: 0 }
2019-01-30 10:04:21.290 - DEBUG - opcua_client::comms::tcp_transport - Connected
2019-01-30 10:04:21.290 - DEBUG - opcua_client::session - open_secure_channel
2019-01-30 10:04:31.284 - INFO - opcua_client::session_state - Timeout waiting for response from server
2019-01-30 10:04:31.284 - INFO - opcua_client::message_queue - Request 56 has timed out and any response will be ignored
2019-01-30 10:04:31.284 - DEBUG - opcua_client::session_state - async_publish, request sent with handle 55
2019-01-30 10:04:31.284 - DEBUG - opcua_client::message_queue - Request 55 was processed by the server
2019-01-30 10:04:31.284 - INFO - opcua_client::session_state - Making secure channel request
2019-01-30 10:04:31.284 - INFO - opcua_client::session_state - security_mode = None
2019-01-30 10:04:31.285 - INFO - opcua_client::session_state - security_policy = None
2019-01-30 10:04:31.285 - DEBUG - opcua_core::comms::secure_channel - AsymmetricSecurityHeader = AsymmetricSecurityHeader { security_policy_uri: UAString { value: Some("http://opcfoundation.org/UA/SecurityPolicy#None") }, sender_certificate: ByteString { value: None }, receiver_certificate_thumbprint: ByteString { value: None } }
2019-01-30 10:04:31.285 - DEBUG - opcua_client::message_queue - Request 57 was processed by the server
2019-01-30 10:04:31.291 - ERROR - opcua_core::comms::chunker - Secure channel id 6 does not match expected id 7
```
I think this is due to the fact that the read state is still running and not dropping. I'll investigate this a little further.

locka99, 2019-01-30:
I made some progress on the feature last night. The code sets the connection state to finished if there is a read/write error and also kicks into the session retry behaviour. However it is not reestablishing subscriptions yet. I've written a TransferSubscriptions stub for the client but I have to hook it up to the reconnect_and_activate() function

locka99, 2019-01-30:
I just put a transfer_subscriptions into the reconnect however there are still some issues with subscription which I'll have to relook at tonight. I think it is recreating the subscription but there is an issue with publish.

denizs, 2019-01-30:
Ok nice - where did you put these changes so I can also have a look over them?

locka99, 2019-01-30:
Mostly in client's session.rs, but also tcp_transport.rs. The history from today / yesterday.

https://github.com/locka99/opcua/commits/master

I changed the polling loop to check for finished state and use a session retry policy to reconnect up to the default number of attempts / interval. If retry doesn't work it then drops out. The sample client'spolling loops were changed slightly to break on an error.

This still isn't reestablishing subscriptions after reconnecting. I've only tested against a Node OPC UA server so far, so I don't know the behaviour with other OPC UA servers. My own server impl doesn't support reconnect yet, at least not using TransferSubscriptions.

locka99, 2019-02-01:
The client code now should reconnect and reestablish subscriptions.

I still have some other things to consider, but the basics are working. Remaining work would be to:

- Think how retry policy should work on the initial connection
- I might be spawning too many publish timers, so the code might need to be refined
- Testing the flow where the connection drops to the server but not because it has been restarted but because of a network issue. In that case, the new connect should be able to activate / resume its existing session.

locka99, 2019-02-01:
And one other piece of work:

* Stop the reconnect from happening if there is an explicit disconnect / abort on the client.

locka99, 2019-02-02:
I've updated connect() to also use the session reconnect policy associated with the session and added a setter to modify what that policy is.

So at present connect() will use the retry policy and if connection is dropped, the standard polling loop will attempt to reconnect using it too. The issue of too many publish timers is still TODO, as is explicit disconnect() calls.

denizs, 2019-02-13:
Sorry for the wait!
Tested your implementation (awesome work btw🎉 ) and discovered that - at least for my (commercial) server implementation - in case the `activate_session` call returns an `Ok(())`, my subscriptions are lost and the subscription timer finishes. #23 addresses this by calling `transfer_subscriptions_from_old_session` in any case. This has been tested in prod and solves all connection issues I had so far 👍

locka99, 2019-02-13:
Thanks I've merged the change and I'll close this out.

closed 2019-02-13 by locka99