locka99/opcua #24: Question about ARM cross-compile

biancode, 2019-02-14:
Hi Adam,

are you experienced to cross-compile for ARM or ARMv7, please?

I tried to run `cargo build --target=arm-unknown-linux-gnueabi` or `cargo build --target=armv7-unknown-linux-gnueabihf` , but it does not compile crates with openssl dependency.
Some advice would very great. I searched for the environment variables or PATH entries, but it doesn't compile.

Happy coding!
Klaus

biancode, 2019-02-14:
state of code is latest master from today

skopinec, 2019-02-14:
This worked for me (Ubuntu 18.04):
1) Install cross-compile utilities as shown here: https://github.com/sodiumoxide/sodiumoxide
2) Follow the malbarbo's answer here: https://stackoverflow.com/questions/37375712/cross-compile-rust-openssl-for-raspberry-pi-2

After compiling the openssl and setting the env. variables, Adam's OPC UA server cross-compiles (and works) just fine.

locka99, 2019-02-14:
Thanks for the hints, I might put this bug # into a readme. I haven't cross-compiled any Rust before but I recognize that OpenSSL is a painpoint. Unfortunately I don't know of any pure Rust crypto library with all the sym / asym crypto, hashing functions used by OPC UA that would allow it to be replaced.

In general cross-compiling requires you to have a full toolchain for the target device plus any headers, static libs etc. you need. A lot of people have used Yocto tools to build custom raspberry pi images and there is advice for getting Rust going in this blog - https://pagefault.blog/2018/07/04/embedded-development-with-yocto-and-rust/

locka99, 2019-02-16:
I've created a doc - https://github.com/locka99/opcua/blob/master/docs/cross-compile.md for cross compile. It appears to work for Raspberry Pi but I haven't tested extensively to see if there are any runtime issues in opcua.

closed 2019-02-16 by locka99