locka99/opcua #27: Revised session timeout is not taken into account

ColinFinck, 2019-02-26:
When creating a new session and requesting a session timeout, the server automatically adjusts this timeout and reports it back to the client. The opcua-client crate picks it up as the `revised_session_timeout` variable in `CreateSessionResponse`, but doesn't do anything with it.
It neither starts a keepalive thread to prevent sessions from timing out (as e.g. [python-opcua is doing](https://github.com/FreeOpcUa/python-opcua/blob/7c6985288b212ae8bc5fd1c8604a05ffa3d9538d/opcua/client/client.py#L29)), nor it propagates this value down to the caller to let it prevent timeouts itself.

The workaround from PR #26 easily solves this problem for us for the moment.
However, in the long term, opcua should be fixed to handle this issue reliably.

locka99, 2019-02-26:
I'll take a look at this.

locka99, 2019-02-27:
I've added a little debugging to the create session for the time being. I'm thinking of putting the requested timeout under the control of the clientbuilder so it can be set and storing the revised timeout in the session. But its more of a hint for the server - the client still has to call the server more frequently than the timeout, e.g. by subscribing to something with a lower interval.

denizs, 2019-02-27:
This would make up for a really nice feature, if one could configure the `CreateSessionRequest` from the `ClientBuilder`. Let us know, if @ColinFinck and I can help.

locka99, 2019-02-27:
Probably the easiest way to make the value controllable would be to add a session_timeout field to the existing SessionRetryPolicy which is passed when creating the session. The SessionRetryPolicy is built using information supplied by from the builder already, so it should be fairly easy to add an extra value.

Slightly harder would be to produce a keep-alive action akin to what python is doing. It would probably have to be a tokio timer task, similar to the ones in TcpTransport but at a higher level. It would have to be spawned from inside create_session after the revised timeout comes back and while the session was in a running state do something like the python-opcua does - make a throwaway request to the server. There is potential for this cause problems however, e.g. if connection was down when the timer fired then the task shouldn't attempt to reconnect or do anything re-entrant. It should just fail. The task also needs to reset if create_session() is called twice. I'll think how it might be achieved. The workaround for the time being would be to create a throwaway subscription with an interval less than the session timeout and just ignore it.

locka99, 2019-02-27:
I've added code to make the session timeout configurable from the builder / config. By default if you set it to 0, it will be the default which is still f64::max. The config / builder take a u32 since I don't see the point of exposing f64 into this part of the code but it is coerced in and out of a f64 during create session.

I haven't put code in yet that will ping the server. I may have time to look at that tomorrow, but it would be similar to my description above - some kind of tokio timer that runs faster than the session timeout and does something to keep the connection alive.

locka99, 2019-03-02:
I've implemented a session activity task. The implementation will send an empty read request to the server at an interval of 0.75 * session timeout.

closed 2019-03-02 by locka99