locka99/opcua #30: Comparison of OPC UA implementations?

shrmrf, 2019-03-21:
Hello @locka99 

Have you done a comparison of this implementation with the Open62541 implementation of OPC UA?

I was just curious to know what's good, what's bad... Are you using this implementation of OPC-UA in production somewhere? I'm looking at different implementations.

Thanks

locka99, 2019-03-22:
I mostly use Node OPC UA to independently test my client / server - I have a server.js and client.js that correspond to simple-server and simple-client. I also connect to my server using free to download binary clients - Softing OPC and UA Expert with no issue. 

I have looked at other open source implementations of OPC UA but I have no interopability samples. Glancing at the Open62541 docs it seems to be a similar product featurewise and is aiming for embedded profile.

CTT is something I'd like to be able to look but I need to be a corporate member of the OPC UA Foundation to get the tools. I do have quite a substantial number of unit tests, and I'm working on some integration tests.

Finally, I don't use this implementation in production. I have tested the client and server against commercial products but I don't vouch for their compliance or longevity. I'm expecting to drop a 0.6 version soon and 0.7 will involve backfilling and more testing. Hopefully as the version number increases, so too will compliance and performance.

closed 2019-03-22 by locka99

shrmrf, 2019-03-22:
Thanks @locka99 for the answer!!

This was a good overview and thank you for the details.

shrmrf, 2019-03-23:
I just added this repository to the list here: https://github.com/open62541/open62541/wiki/List-of-Open-Source-OPC-UA-Implementations