locka99/opcua #32: tokio task is not stopped if session is dropped

denizs, 2019-05-17:
In our setup, we are experiencing unpredictable behaviour of the OPC server where sessions are dropped silently on server side, forcing us to close and reconnect as soon as a read out fails. 
Our implementation involves dropping the current session and starting from scratch.

If this happens a couple of times, we are running in to "Too many open files", because the `connection_task` run by Tokio is not terminated after a session is dropped.

locka99, 2019-05-20:
This is most likely the same as #16

denizs, 2019-05-20:
Do you have any clue on how to fix this? Looking at the code from a high level perspective, it seems like properly cleaning up the `connection_task` should do the trick, or am I mistaking

locka99, 2019-05-20:
I'll have to go back and see if I can debug it again. Part of the reason all the tasks have debugging around their success / failure states is from previous attempts to figure out which task wasn't exiting when it was supposed to.

If your server is trying to periodically register with an OPC discovery server and this is the cause, then one workaround is to disable that behaviour. 

e.g. i.e. set this via a builder

builder.discovery_server_url(None)

or this in a server.conf

discovery_server_url: ~

Then the server won't periodically call the discovery server, which it does an OPC UA client and therefore there is no call to create a connection_task.

denizs, 2019-05-30:
The problem is that we don't have any control over the server. We are using the client only.

denizs, 2019-06-03:
I investigated this issue a little bit more. If you call disconnect, the `WriteState` isn't dropped. This is easy to achieve by calling `close_secure_channel` on the session. However, this doesn't seem to close the open file descriptor. I then had a look into the `WriteHalf<TcpStream>` and found that apparently, the `shutdown` method [doesn't do anything](https://github.com/tokio-rs/tokio/issues/852).

locka99, 2019-06-03:
Thanks for investigating this. I'll see if I can spare some time to look I to this. I know I've wrapped some futures in a completion pact struct (https://github.com/locka99/opcua/blob/master/server/src/completion_pact.rs) to put additional logic in and perhaps the same can be done here. Obviously if I can't stop a task when it is an unrecoverable state then that would be bad....

denizs, 2019-06-05:
It seems like this is indeed related to #16. When letting the process continue, I see debug output of tokio_reactor::registration scheduling reads. Seems like the read half stays open

locka99, 2019-09-11:
I think I've fixed this on the trunk. There are a number of scenarios where some of the tasks in the client / server side wouldn't necessarily close properly and I've tried to make it more robust. The reader task now sends an explicit quit to the writer when the socket breaks or finishes with an error.

closed 2019-09-11 by locka99