locka99/opcua #34: New events on a subscription are not resetting the keep alive so the subscription expires

locka99, 2019-09-15:
The subscriptions code needs to correctly reset the keep alive on the sub when an monitored event happens.

pulzzedavid, 2020-05-25:
Any work around for this issue?
We may be hitting this issue where if data is available at around the publish interval, it seems to work fine, but when data is available slower than publish interval, subscription is killed after around couple of minutes.

locka99, 2020-05-26:
Subscriptions are managed by a state table which dictates how long they are kept alive for depending on there being pending notifications, publish requests etc.

It is very complex, especially when trying to work from the docs and it is possible that something I am doing is not intended by the specification. 

You may also find that if you set your own requestedLifetimeCount and requestedMaxKeepAliveCount higher when you call create_subscription that you are able to sustain the subscription for longer.

pulzzedavid, 2020-05-27:
No doubt it is a complex logic. Thanks for sharing the details. Let me debug it further and see if I can pinpoint the issue.