locka99/opcua #35: cannot build on windows with nightly-x86_64-pc-windows-msvc

nkbai, 2019-09-16:
it depends on  signal-hook-registry,but this library only works on unix.
```
 Compiling signal-hook-registry v1.0.1
error[E0432]: unresolved imports `libc::sigaction`, `libc::siginfo_t`, `libc::sigset_t`, `libc::SIG_BLOCK`, `libc::SIG_SETMASK`
  --> C:\Users\bai\.cargo\registry\src\github.com-1ecc6299db9ec823\signal-hook-registry-1.0.1\src\lib.rs:58:27
   |
58 | use libc::{c_int, c_void, sigaction, siginfo_t, sigset_t, SIG_BLOCK, SIG_SETMASK};
   |                           ^^^^^^^^^  ^^^^^^^^^  ^^^^^^^^  ^^^^^^^^^  ^^^^^^^^^^^ no `SIG_SETMASK` in the root
   |                           |          |          |         |
   |                           |          |          |         no `SIG_BLOCK` in the root
   |                           |          |          no `sigset_t` in the root
   |                           |          no `siginfo_t` in the root
   |                           no `sigaction` in the root

error[E0432]: unresolved imports `libc::SIGKILL`, `libc::SIGSTOP`
  --> C:\Users\bai\.cargo\registry\src\github.com-1ecc6299db9ec823\signal-hook-registry-1.0.1\src\lib.rs:60:28
   |
60 | use libc::{SIGFPE, SIGILL, SIGKILL, SIGSEGV, SIGSTOP};
   |                            ^^^^^^^           ^^^^^^^ no `SIGSTOP` in the root
   |                            |
   |                            no `SIGKILL` in the root
   |                            help: a similar name exists in the module: `SIGILL`
```

locka99, 2019-09-16:
I usually only build with stable-x86_64-pc-windows-gnu (1.37) but I built there with stable-x86_64-pc-windows-msvc and aside from setting OPENSSL_DIR env to point to a OpenSSL MSVC binaries, I was able to build successfully.

locka99, 2019-09-28:
1.38 also works for me which I assume to be the nightly when you had the problem so I am closing this. My attempt to reproduce

set OPENSSL_DIR=C:\OpenSSL-Win64
rustup default stable-msvc
cargo build

I was also able to run the demo-server without issue.

closed 2019-09-28 by locka99