locka99/opcua #38: How to retrieve NamespaceIndexes with the client

wofr, 2019-10-07:
Hi

Is there a way for the OPC-UA client to retrieve (browse) the name-space Indexes of an server (root-folder)

I tried the browse function of the  
```
let bd:[BrowseDescription;1] = [BrowseDescription {
            node_id: NodeId::new(0,"RootFolder"),
            browse_direction: BrowseDirection::Forward,
            reference_type_id: NodeId::new(0,"RootFolder"),
            include_subtypes: true,
            node_class_mask:  0xff,
            result_mask:  0xff
        }];

match session.browse( &bd) {
            Ok(r) => {
                let e = r.unwrap();
                for i in &e{
                    println!("Status Code {:?} ",i.status_code);
                    //println!("Status Code {:?} {:?}",i.continuation_point,i.status_code);
                }
            }
            Err(e)=> println!("Faied due to {:?}",e.to_string()),
        };
```

But this always returns
`Status Code IS_ERROR | BadResourceUnavailable | BadEncodingLimitsExceeded | BadShutdown | BadNotImplemented | BadMonitoredItemFilterUnsupported | BadContentFilterInvalid | BadReferenceTypeIdInvalid`

The session it-self must be valid because creating a monitored item subscription with hard-coded namespaceIdx works fine.

Best Regards

Wolfgang

locka99, 2019-10-07:
It'll be easier to read the response code if you use {} instead of {:?}. The bitflags macro messes up the response for StatusCode in debug mode, i.e. "println!("Status Code {} ",i.status_code);"

The Root folder's node id is not ns=0;s=RootFolder. It is ns=0;i=84 and there are 3 ways you could make it:

```
let node_id = NodeId::new(0, 84); // Non-obvious
let node_id: NodeId = ObjectId::RootFolder.into();
let node_id = NodeId::root_folder_id();
```

The reference_type_id should be something like Organizes, e.g. 

`let reference_type_id: NodeId = ReferenceTypeId::Organizes.into();`

wofr, 2019-10-07:
Thanks that solved my problem.

closed 2019-10-07 by wofr