locka99/opcua #41: Feature Request: example of writing to a variable

dunmatt, 2020-01-10:
Hi!  First of all, thanks for writing this library, I'm excited to see it work.

I'm having some trouble writing to variables in a server running on a PLC.  I know it's on the client side because I can write to them just fine via UaExpert... but for the life of me I can't figure out how to write to it via this library.  My code is 

    let mut client = ClientBuilder::new()
        .application_name("Vicarious RSI Driver")
        .application_uri("urn:vicarious/kuka/driver")
        .trust_server_certs(true)
        .session_retry_limit(3)
        .endpoint("only_endpoint", ClientEndpoint {
            url: String::from(url),
            security_policy: SecurityPolicy::None.to_string(),
            security_mode: MessageSecurityMode::None.to_string(),
            user_token_id: ANONYMOUS_USER_TOKEN_ID.to_string(),
        })
        .default_endpoint("only_endpoint")
        .client().ok_or("Couldn't build an OPC client???")?;
    let session = client.connect_to_endpoint_id(Some("only_endpoint")).map_err(|e| e.to_string())?;

    if let Ok(mut session) = session.write() { // locks the rw lock
        let request = NodeId::new(6, "::AsGlobalPV:IO.VicStack1.Dinp.blnEnableRq").into();
        println!("{:?}\n", session.read(&[request]));

        let request = WriteValue {
            node_id: NodeId::new(6, "::AsGlobalPV:IO.VicStack1.Dinp.blnEnableRq"),
            attribute_id: 13,
            index_range: UAString { value: None },
            value: DataValue::new(false),
        };
        println!("{:?}\n", session.write(&[request]));
    }
    let _ = Session::run(session);
    println!("done");

and it yields:

    Ok(Some([DataValue { value: Some(Boolean(true)), status: None, source_timestamp: None, source_picoseconds: None, server_timestamp: Some(DateTime { date_time: 2000-03-06T05:03:59.298Z }), server_picoseconds: None }]))
    
    Ok(Some([IS_ERROR | GoodClamped | BadUnexpectedError | BadInternalError | BadOutOfMemory | BadTooManyOperations | BadDataTypeIdUnknown | BadCertificateInvalid | BadSecurityChecksFailed | BadIdentityTokenInvalid | BadIdentityTokenRejected | BadSecureChannelIdInvalid | BadInvalidTimestamp | BadNoCommunication | BadWaitingForInitialData | BadNodeIdInvalid | BadNotImplemented | BadMonitoringModeInvalid | BadMonitoredItemIdInvalid | BadMonitoredItemFilterInvalid | BadServerNameMissing | BadDiscoveryUrlMissing | BadSempahoreFileMissing | BadRequestTypeInvalid | BadBrowseNameInvalid | BadBrowseNameDuplicated | BadNodeAttributesInvalid | BadTypeDefinitionInvalid | BadMaxAgeInvalid | BadHistoryOperationInvalid | BadHistoryOperationUnsupported | BadWriteNotSupported]))

It seems unlikely that I'm really earning every possible error, and more likely that I'm botching something simple, but I'm not getting traction figuring out what.  Could I persuade you to throw together a working example of a client updating a value on a server?

locka99, 2020-01-14:
Sorry for not responding sooner, but use {} not {:?} when debugging the StatusCode. I'm using the bitflags crate for StatusCode and it messes up the output otherwise.

The error code might yield a useful explanation then.

Are you using the 0.8 release or the master branch? As far as I can tell your code appears to be setting a Boolean value into to a node id in the correct manner. You probably don't need the Session::run() immediately following it however if you intend to exit immediately.

dunmatt, 2020-01-22:
Thanks for the advice, it seems I'm getting a BadWriteNotSupported.  As for the version, I'm on 0.7.0 from crates.io.  Could that be related?

dunmatt, 2020-01-25:
Solved it!!  The problem was that `DataValue::new(true)` populated the source timestamp fields, and for whatever reason that was causing my PLC to reject the update.  I suspect that it's because the PLC's clock is 20+ years slow, but our PLC guy insists that that's not the reason.  I'm not going to argue over it.

locka99, 2020-01-25:
That's great to know!

The workaround and to make it explicit would be to say:

value: DataValue::value_only(true)

Or

value: DataValue {
            value: Some(false.into()),
            status: Some(StatusCode::Good),
            source_timestamp: None,
            source_picoseconds: None,
            server_timestamp: None,
            server_picoseconds: None,
}

closed 2020-01-25 by locka99

locka99, 2020-01-25:
I just looked up the spec and there is an explicit mention of this that says for the Write service - "If the SourceTimestamp or the ServerTimestamp is specified, the Server shall use these values. The Server returns a Bad_WriteNotSupported error if it does not support writing of timestamps"