locka99/opcua #43: "nonce must be non-zero" when trying to connect to OPC UA server from milo

evanjs, 2020-02-19:
When attempting to connect to an OPC UA server configured with `ServerConfig::load` from a milo client, I am unable to connect to the server, and receive [this](https://github.com/eclipse/milo/blob/35ee06ea87676db9fdf4caf748b9265c40d1c841/opc-ua-stack/stack-core/src/main/java/org/eclipse/milo/opcua/stack/core/util/NonceUtil.java#L184) error message: "nonce must be non-zero".

According to [this comment](https://github.com/eclipse/milo/issues/575#issuecomment-558373517), I need to set the nonce whether security is enabled or not?
From what I've found in the samples, this can be done in the code, but I'm unsure how I can achieve this from a server configuration file.

Am I missing something with my configuration? Or is there something I need to do on the rust side?

<details>
 <summary>Current server.conf</summary>

``` 
---
application_name: OPC UA Sample Server
application_uri: "urn:OPC UA Sample Server"
product_uri: ""
pki_dir: "./pki"
create_sample_keypair: true
trust_client_certs: false
discovery_server_url: "opc.tcp://localhost:4840/UADiscovery"
tcp_config:
  hello_timeout: 5
  host: 0.0.0.0
  port: 4855
limits:
  clients_can_modify_address_space: false
  max_subscriptions: 100
  max_monitored_items_per_sub: 1000
  max_array_length: 1000
  max_string_length: 65536
  max_byte_string_length: 65536
  min_sampling_interval: 0.1
  min_publishing_interval: 0.1
user_tokens:
  sample_password_user:
    user: sample
    pass: sample1
  sample_x509_user:
    user: sample_x509
    x509: "./users/sample-x509.der"
  unused_user:
    user: unused
    pass: unused1
discovery_urls:
  - "opc.tcp://127.0.0.1:4855/"
endpoints:
  basic128rsa15_sign:
    path: /
    security_policy: Basic128Rsa15
    security_mode: Sign
    security_level: 2
    password_security_policy: ~
    user_token_ids:
      - ANONYMOUS
      - sample_password_user
      - sample_x509_user
  basic128rsa15_sign_encrypt:
    path: /
    security_policy: Basic128Rsa15
    security_mode: SignAndEncrypt
    security_level: 2
    password_security_policy: ~
    user_token_ids:
      - ANONYMOUS
      - sample_password_user
      - sample_x509_user
  basic256_sign:
    path: /
    security_policy: Basic256
    security_mode: Sign
    security_level: 3
    password_security_policy: ~
    user_token_ids:
      - ANONYMOUS
      - sample_password_user
      - sample_x509_user
  basic256_sign_encrypt:
    path: /
    security_policy: Basic256
    security_mode: SignAndEncrypt
    security_level: 3
    password_security_policy: ~
    user_token_ids:
      - ANONYMOUS
      - sample_password_user
      - sample_x509_user
  basic256sha256_sign:
    path: /
    security_policy: Basic256Sha256
    security_mode: Sign
    security_level: 4
    password_security_policy: ~
    user_token_ids:
      - ANONYMOUS
      - sample_password_user
      - sample_x509_user
  basic256sha256_sign_encrypt:
    path: /
    security_policy: Basic256Sha256
    security_mode: SignAndEncrypt
    security_level: 4
    password_security_policy: ~
    user_token_ids:
      - ANONYMOUS
      - sample_password_user
      - sample_x509_user
  no_access:
    path: /noaccess
    security_policy: None
    security_mode: None
    security_level: 1
    password_security_policy: ~
    user_token_ids: []
  none:
    path: /
    security_policy: None
    security_mode: None
    security_level: 1
    password_security_policy: ~
    user_token_ids:
      - ANONYMOUS
      - sample_password_user
      - sample_x509_user
```

</details>

Thanks!

locka99, 2020-02-20:
Hi, can you describe what security profiles you're running on the server and if you have a server certificate etc. In addition what security profile the client is trying to create a session with on the server and if the client also supplies a certificate.

There is nothing you as a caller to the API should do about this. The general handshake mechanism is the server and client exchange certificates and then for each to sign a randomly generated nonce to prove they are in possession of their respective private keys. Then the nonce is used to create a session key.

But if one side is not supplying a nonce or there is some mismatch between cert / nonce / profile on either the client or server side then it might break. I can't say for sure if this is a bug in the milo code or in the rust code.

If you can attach logs for both client / server we might be able to see which side didn't like what it got from the other.

locka99, 2020-02-20:
One thing I note from looking at the milo code is "must be non-zero" error is not checking if the nonce is zero length, but if it contains all zeroes. 

https://github.com/eclipse/milo/blob/35ee06ea87676db9fdf4caf748b9265c40d1c841/opc-ua-stack/stack-core/src/main/java/org/eclipse/milo/opcua/stack/core/util/NonceUtil.java#L184

On my code, I only bother to create a random nonce if the security policy is NOT None otherwise I hand out a 1 byte placeholder nonce

https://github.com/locka99/opcua/blob/7859c1df3265da8f7038d14e303bd526aa1c80c5/core/src/comms/secure_channel.rs#L254

I am guessing that function argument minimumLength is 0 or 1 so it doesn't bother to flag the length of the nonce, but seeing it be more than 0 means it then checks for zeroes and fires the spurious error.

I do not recall why I return a 1 byte nonce but I may have had issues with passing an empty or zero length nonce even though the field is not used when the security policy is None. As a hack, if you have my source you can modify that create_random_nonce() to assign vec![1u8; 1] instead of vec![0u8; 1] just to pass the test. It might also work to assign a Vec::new() too.

evanjs, 2020-02-21:
Disclaimer: This is an issue a coworker is experiencing so pardon anything I'm misunderstanding or missing.

Okay, so after forcing `create_random_nonce` to return `vec![1u8; 1];`, milo throws `Error running example: java.lang.Exception: no anonymous token policy found`.

Note that this is while running [this](https://github.com/eclipse/milo/blob/5c6048d54be28449358f0a395bd8b02924127c51/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client/BrowseNodeExample.java#L24) example with [ClientExample.getEndpointUrl](https://github.com/eclipse/milo/blob/5ed4edad57be181b3b0d08fb124e252cf2decc56/milo-examples/client-examples/src/main/java/org/eclipse/milo/examples/client/ClientExample.java#L26) set to return `opc.tcp://localhost:4855/`.

<details>
<summary>Relevant logs from my opc ua server</summary>

```
2020-02-21 14:36:56.599 - ERROR - opcua_server::comms::tcp_transport - Server reader session status is BadConnectionClosed so finishing
2020-02-21 14:36:56.600 - ERROR - opcua_server::comms::tcp_transport - Server reader error Kind(ConnectionReset)
2020-02-21 14:36:56.600 - ERROR - opcua_server::comms::tcp_transport - Read loop is finished with an error ()
2020-02-21 14:36:56.675 - ERROR - opcua_server::comms::tcp_transport - Server reader session status is BadConnectionClosed so finishing
2020-02-21 14:36:56.675 - ERROR - opcua_server::comms::tcp_transport - Server reader error Kind(ConnectionReset)
2020-02-21 14:36:56.675 - ERROR - opcua_server::comms::tcp_transport - Read loop is finished with an error ()
```

</details>

<details>
<summary>Relevant logs from milo's BrowseNodeExample (With non-zero nonce) </summary>

```
09:36:54,728 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Could NOT find resource [logback.groovy]
09:36:54,729 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Could NOT find resource [logback-test.xml]
09:36:54,729 |-INFO in ch.qos.logback.classic.LoggerContext[default] - Found resource [logback.xml] at [file:/home/evanjs/src/milo/milo-examples/client-examples/target/classes/logback.xml]
09:36:54,730 |-WARN in ch.qos.logback.classic.LoggerContext[default] - Resource [logback.xml] occurs multiple times on the classpath.
09:36:54,730 |-WARN in ch.qos.logback.classic.LoggerContext[default] - Resource [logback.xml] occurs at [file:/home/evanjs/src/milo/milo-examples/client-examples/target/classes/logback.xml]
09:36:54,730 |-WARN in ch.qos.logback.classic.LoggerContext[default] - Resource [logback.xml] occurs at [file:/home/evanjs/src/milo/milo-examples/server-examples/target/classes/logback.xml]
09:36:54,787 |-INFO in ch.qos.logback.classic.joran.action.ConfigurationAction - debug attribute not set
09:36:54,792 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - About to instantiate appender of type [ch.qos.logback.core.ConsoleAppender]
09:36:54,793 |-INFO in ch.qos.logback.core.joran.action.AppenderAction - Naming appender as [STDOUT]
09:36:54,834 |-WARN in ch.qos.logback.core.ConsoleAppender[STDOUT] - This appender no longer admits a layout as a sub-component, set an encoder instead.
09:36:54,834 |-WARN in ch.qos.logback.core.ConsoleAppender[STDOUT] - To ensure compatibility, wrapping your layout in LayoutWrappingEncoder.
09:36:54,834 |-WARN in ch.qos.logback.core.ConsoleAppender[STDOUT] - See also http://logback.qos.ch/codes.html#layoutInsteadOfEncoder for details
09:36:54,834 |-INFO in ch.qos.logback.classic.joran.action.RootLoggerAction - Setting level of ROOT logger to INFO
09:36:54,834 |-INFO in ch.qos.logback.core.joran.action.AppenderRefAction - Attaching appender named [STDOUT] to Logger[ROOT]
09:36:54,835 |-INFO in ch.qos.logback.classic.joran.action.ConfigurationAction - End of configuration.
09:36:54,835 |-INFO in ch.qos.logback.classic.joran.JoranConfigurator@7c0c77c7 - Registering current configuration as safe fallback point

09:36:55.033 [main] INFO  o.e.m.examples.server.ExampleServer - security temp dir: /tmp/security
09:36:55.038 [main] INFO  o.e.m.examples.server.KeyStoreLoader - Loading KeyStore at /tmp/security/example-server.pfx
09:36:55.157 [main] INFO  o.e.m.examples.server.ExampleServer - pki dir: /tmp/security/pki
09:36:55.665 [main] INFO  o.e.m.opcua.sdk.server.OpcUaServer - Eclipse Milo OPC UA Stack version: dev
09:36:55.666 [main] INFO  o.e.m.opcua.sdk.server.OpcUaServer - Eclipse Milo OPC UA Server SDK version: dev
09:36:56.227 [main] INFO  o.e.m.o.s.s.n.OpcUaNamespace - Loaded nodes in 351ms.
09:36:56.328 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://sekka:12686/milo to 0.0.0.0:12686 [None/None]
09:36:56.373 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://sekka:12686/milo to 0.0.0.0:12686 [Basic256Sha256/SignAndEncrypt]
09:36:56.373 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://sekka:12686/milo/discovery to 0.0.0.0:12686 [None/None]
09:36:56.373 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://localhost:12686/milo to 0.0.0.0:12686 [None/None]
09:36:56.373 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://localhost:12686/milo to 0.0.0.0:12686 [Basic256Sha256/SignAndEncrypt]
09:36:56.373 [main] INFO  o.e.m.o.stack.server.UaStackServer - Binding endpoint opc.tcp://localhost:12686/milo/discovery to 0.0.0.0:12686 [None/None]
[... cut ...]
	at io.netty.util.HashedWheelTimer$HashedWheelBucket.expireTimeouts(HashedWheelTimer.java:757)
	at io.netty.util.HashedWheelTimer$Worker.run(HashedWheelTimer.java:485)
	at java.base/java.lang.Thread.run(Thread.java:834)

Process finished with exit code 0

```

</details>

Regarding certificates, it sounds like we're currently using a self-signed certificate for the server.

Please let me know if I need to provide any more logs with e.g. `RUST_OPCUA_LOG=debug` or etc, or if there's any more information I can provide.

locka99, 2020-02-21:
Writing this from a mobile so quick request - can you try the Vec::new() there too?

evanjs, 2020-02-21:
Sure.
Looks like I still get `Error running client example: java.lang.Exception: no anonymous token policy found`.

evanjs, 2020-02-21:
Okay, going back to milo, it looks like this is where it's failing.

``` java
public class AnonymousProvider implements IdentityProvider {

    public static final IdentityProvider INSTANCE = new AnonymousProvider(); 

    @Override
    public SignedIdentityToken getIdentityToken(
        EndpointDescription endpoint,
        ByteString serverNonce) throws Exception {

        List<UserTokenPolicy> userIdentityTokens = l(endpoint.getUserIdentityTokens());

        return userIdentityTokens.stream()
            .filter(t -> t.getTokenType() == UserTokenType.Anonymous)
            .findFirst()
            .map(policy -> {
                UserIdentityToken token = new AnonymousIdentityToken(policy.getPolicyId());

                return new SignedIdentityToken(token, new SignatureData());
            })
            .orElseThrow(() -> new Exception("no anonymous token policy found"));
    }

    @Override
    public String toString() {
        return "AnonymousProvider{}";
    }

}
```

It seems like it's trying to find user tokens with a token type of `Anonymous` in an empty list of UserTokenPolicies.

evanjs, 2020-02-22:
At any rate, I believe this particular issue can be closed, as the nonce-related behavior seems to be solved by 2f6a662.  Thank you!

closed 2020-02-22 by evanjs