locka99/opcua #44: Panic when running sample servers

Troels51, 2020-02-20:
When running the any of the sample servers, and then using any of the sample clients i get this panic on the server.
```
thread 'tokio-runtime-worker-2' panicked at 'explicit panic', server/src/state.rs:145:20
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace.
thread 'tokio-runtime-worker-5' panicked at 'called `Result::unwrap()` on an `Err` value: "PoisonError { inner: .. }"', src/libcore/result.rs:thread 'thread 'tokio-runtime-worker-2tokio-runtime-worker-1' panicked at '' panicked at 'called `Result::unwrap()` on an `Err` value: "PoisonError { inner: .. }"called `Result::unwrap()` on an `Err` value: "PoisonError { inner: .. }"', ', src/libcore/result.rssrc/libcore/result.rs::11651165::55
```
I tracked it down to [user_pass_security_policy_id](https://github.com/locka99/opcua/blob/8ac59f0c616d1df8cfa150c63f38062a5cdb3466/server/src/state.rs#L140)
Which only matches a subset of the SecurityPolicies.

I worked around it by just adding a couple of new cases, but i'm not sure what the correct behaviour is, and maybe it should fail but with a better error description.

evanjs, 2020-02-21:
Also experienced this when trying to connect with the the [milo](https://github.com/eclipse/milo/) client.

locka99, 2020-02-21:
I'm on a mobile so just a quick request - can you try with Vec::new() as well?

locka99, 2020-02-21:
Sorry writing to wrong bug

locka99, 2020-02-21:
I've been adding a couple of new security policies these last few weeks and the work isn't complete yet. From the looks of it, it is triggering a panic because the new policies are falling into the default case. To counter this, I've just put a placeholder in to user_pass_security_policy_id() to stop that happening.

locka99, 2020-02-23:
I'll close this since I don't observe a panic after my recent change. However I'm still building out the 2 new security profiles so the change is it a stopgap

closed 2020-02-23 by locka99