locka99/opcua #45: Unable to connect to server anonymously

evanjs, 2020-02-22:
Split from #43 
___
When attempting to connect to an OPC UA server configured with `ServerConfig::load` from a milo client, I am unable to connect to the server, and receive [this](https://github.com/eclipse/milo/blob/master/opc-ua-sdk/sdk-client/src/main/java/org/eclipse/milo/opcua/sdk/client/api/identity/AnonymousProvider.java#L47) error message: `no anonymous token policy found`.
___
I found one way to work around this issue, though I'm not entirely sure why it's happening.

I am able to connect if I specify an additional endpoint in the `server.conf`, whether or not it is connected to.

Example:
``` yaml
  milo:
    path: /milo
    security_policy: None
    security_mode: None
    security_level: 1
    password_security_policy: ~
    user_token_ids:
      - ANONYMOUS
      - sample_password_user
      - sample_x509_user
```
___
To recap:
With the default `server.conf`, I am unable to connect to the opcua server anonymously, as the client cites it can't find any anonymous token policies.
However, if I specify seemingly any additional (i.e. not `/`) endpoints in the `server.conf`, I can then connect to the root endpoint anonymously.

locka99, 2020-02-23:
Hi, when you say the default server.conf, do you mean the sample server.conf? 

I have just gotten milo built and installed here. I had to edit the BrowseNodeExample to connect to my server. I notice that for some reason it is resolving to the /noaccess endpoint in the sample.conf which is an endpoint that deliberately supports no users as part of its purpose. 

If I edit sample.conf to remove /noaccess the sample works. But if I leave it as-is or even reorder it to be last in the file it fails. 

I think this is because ClientExampleRunner.java line 96 has this code to set the endpoint to use -

```
        EndpointDescription endpoint = endpoints.stream()
            .filter(e -> e.getSecurityPolicyUri().equals(securityPolicy.getUri()))
            .filter(clientExample.endpointFilter())
            .findFirst()
            .orElseThrow(() -> new Exception("no desired endpoints returned"));

```

It looks for the first endpoint to match the security policy without checking if the path matches too. This seems odd to me since the endpoint url is explicitly "opc.tcp://localhost:4855/" and there is a "/" path. This would suggest to me that milo is in error.

The problem endpoints for milo are these:

```
no_access:
    path: /noaccess
    security_policy: None
    security_mode: None
    security_level: 1
    password_security_policy: ~
    user_token_ids: []
  none:
    path: /
    security_policy: None
    security_mode: None
    security_level: 1
    password_security_policy: ~
    user_token_ids:
      - ANONYMOUS
      - sample_password_user
      - sample_x509_user
```

The Rust server loads its endpoints into a BTreeMap so the ordering is the alphasorted string key which in this case is "no_access" followed by "none". Even if I write them in a different order they will be returned to the client alpha sorted.

There are a couple of workarounds:

1. Remove no_access since it is an artificial endpoint for testing purposes
2. Rename no_access so it becomes xxx_no_access or similar so it alphasorts differently.
3. Change Rust Server to use some kind of order of insertion map to make ordering more explicitly

Personally though I think the issue is Milo here for not matching the endpoint properly. If I say I want endpoint "/" with None/None and the server provides a "/" with None/None then it shouldn't be choosing another endpoint.

locka99, 2020-02-23:
I think also I'll look at the endpointUrl given to me in the GetEndpointsRequest since I don't filter on that, and perhaps I should. I'll look at the spec again to see if I missed something.

locka99, 2020-02-23:
I read the spec for GetEndpoints() and looked at milo's impl and the endpointUrl shouldn't affect this behaviour. It looks as though I should be more stringent in checking the field so I've updated the code to ensure a valid hostname but I don't think it has an impact on the issue above.

locka99, 2020-02-23:
I think I will check for a bug on the Milo project or raise one of it doesn't exist

evanjs, 2020-02-23:
Thanks! I have confirmed on my side that I am able to run the example after removing the `no_access` endpoint.

Supposedly, a different server that my coworker tested experienced no issues, but perhaps that implementation was simply skirting the issue as well.

locka99, 2020-02-24:
I raised the issue in milo and I think there is general agreement that is just a case of one sample vs another sample resulting in an error so I'll close this out. Milo's example endpoint choosing code is very straightforward and matches the first endpoint with the desired policy and you have to implement a further client endpoint filter to get any more precise than that.

closed 2020-02-24 by locka99