locka99/opcua #48: Security Policy not included in GetEndpointsResponse message

pulzzedavid, 2020-04-12:
First of all, thanks for a great library.

I may be missing some steps, but I am trying to add additional security policies to server without much luck. I have the following.

```
.endpoints(
            [
                ("none", endpoint_path, SecurityPolicy::None, MessageSecurityMode::None, &user_token_ids),
                ("basic128rsa15_sign", endpoint_path, SecurityPolicy::Basic128Rsa15, MessageSecurityMode::Sign, &user_token_ids),
                ("basic128rsa15_sign_encrypt", endpoint_path, SecurityPolicy::Basic128Rsa15, MessageSecurityMode::SignAndEncrypt, &user_token_ids),
                ("basic256_sign", endpoint_path, SecurityPolicy::Basic256, MessageSecurityMode::Sign, &user_token_ids),
                ("basic256_sign_encrypt", endpoint_path, SecurityPolicy::Basic256, MessageSecurityMode::SignAndEncrypt, &user_token_ids),
                ("basic256sha256_sign", endpoint_path, SecurityPolicy::Basic256Sha256, MessageSecurityMode::Sign, &user_token_ids),
                ("basic256sha256_sign_encrypt", endpoint_path, SecurityPolicy::Basic256Sha256, MessageSecurityMode::SignAndEncrypt, &user_token_ids),
            ].iter().map(|v| {
                (v.0.to_string(), ServerEndpoint::from((v.1, v.2, v.3, &v.4[..])))
            }).collect())
        .config();
```

But, when I see the GetEndpointsResponse message, I am only seeing a single endpoint with policy 
```
http://opcfoundation.org/UA/SecurityPolicy#None
```

Wondering whether additional steps are required on top of defining them as above.

locka99, 2020-04-12:
Hi, How are you requesting get endpoints from the client? In particular are you specifying a url, locale ids or profile urls with the request? Does the url match the server's url? Are you verifying the endpoints aren't being overridden by something you do later in the builder, e.g. new_anonymous()?

I can see that if I use ServerBuilder::new_sample().build() to make a demo server that it returns 10 endpoints which is what I would expect. But perhaps there is something in the input which is affecting the output.

pulzzedavid, 2020-04-12:
Happy Easter,

Thanks for getting back to me on this. Yes, looks like it was being overridden. Does trust_client_certs() override the config as well? I am trying to get security policies working, and looks like you have to store trusted certs in specific directory. In order to test the operation, I am trying to set it to not check the certs for trust. But, still getting 0x80130000 (BadSecurityChecksFailed) service fault message even when I use trust_client_certs(). Do I need to set some specific user token ids? I am currently just using anonymous. Some guidance would be great.

pulzzedavid, 2020-04-13:
Got it to work. Looking at the code, even if you use trust_client_certs(), check for cert in rejected folder happens before check for trust_unknown_certs, so it fails.

closed 2020-04-13 by pulzzedavid

locka99, 2020-04-13:
I'm glad you got it sorted. Yes if the cert is in the rejected folder then it might be rejected regardless of throwing the trust switch on the server since it will see it there.

pulzzedavid, 2020-04-13:
Yeah, the name threw me off a little bit. "trust_unknown_certs" makes good sense. But, "trust_client_certs" made me think that it will not check at all. Thanks for the help. Much appreciate it.