locka99/opcua #51: License & Copyright Header is missing in most files

benjaminSchilling33, 2020-05-14:
Hi @locka99,

first of all thanks for your amazing work on this OPC UA implementation in rust.

Unfortunately I'm not feeling very well when using it due to the missing License & Copyright headers in most files.

Would you mind adding one to all files?
If you don't have the time I can also take care of it via a pull request.

If I should do it I propose the following format:

```
opcua
<file name>
SPDX-License-Identifier: MPL-2.0
Copyright (C) 2020 Adam Lock
```

locka99, 2020-05-15:
Leave it with me, I'll add it as a task for the next release

benjaminSchilling33, 2020-05-15:
@locka99 great! Looking forward to it.

locka99, 2020-05-16:
I've added a header now. I'm not going to put the filename into the header, but each file aside from test files has the project name, licence id and copyright notice at the top. I've also updated the scripts that generate these files to insert the message.

closed 2020-05-16 by locka99