locka99/opcua #54: Timeout when connecting to endpoint

dgl2, 2020-05-20:
Hi,

I've been successfully using your library on a past project, but on the current one I've been encountering some strange behaviour during connection:
the software manages to establish a connection to the server and list the available endpoints, but then fails to connect to the endpoint on a timeout.

There seems to be no response from the server past the endpoint listing. Unfortunately I do not have access to server logs, but its operator told me he could not see our connection attempt.

Do you have any clue on what might cause this issue? What information could I ask my client to help debugging?
    
Any help on this issue would be much appreciated.

The server uses a certificate authentication and the problem happens when I call the `client.connect_to_endpoint_id` function.

Here is my (redacted) configuration and the application logs

```
ClientConfig {
    application_name: "XXX",
    application_uri: "xxx",
    create_sample_keypair: false,
    trust_server_certs: true,
    product_uri: "",
    pki_dir: "/tmp/.tmpzeMRNC",
    preferred_locales: [],
    default_endpoint: "test",
    user_tokens: {
        "Foobar": ClientUserToken {
            user: "",
            password: None,
            cert_path: Some(
                "/tmp/.tmpzeMRNC/own/cert.der",
            ),
            private_key_path: Some(
                "/tmp/.tmpzeMRNC/private/private.pem",
            ),
        },
    },
    endpoints: {
        "test": ClientEndpoint {
            url: "opc.tcp://10.1.1.1:1234/DataAccessServer",
            security_policy: "Basic256",
            security_mode: "SignAndEncrypt",
            user_token_id: "Foobar",
        },
    },
    session_retry_limit: 10,
    session_retry_interval: 10000,
    session_timeout: 0,
}
```
```  
13:40:07.18 [main] [opcua/client/src/session.rs:opcua_client::session] [INFO] Connect
13:40:07.18 [main] [opcua/client/src/session.rs:opcua_client::session] [INFO] Security policy = None
13:40:07.18 [main] [opcua/client/src/session.rs:opcua_client::session] [INFO] Security mode = None
13:40:07.38 [main] [opcua/client/src/session_state.rs:opcua_client::session_state] [INFO] Making secure channel request
13:40:07.38 [main] [opcua/client/src/session_state.rs:opcua_client::session_state] [INFO] security_mode = None
13:40:07.38 [main] [opcua/client/src/session_state.rs:opcua_client::session_state] [INFO] security_policy = None
13:40:07.53 [main] [opcua/client/src/session.rs:opcua_client::session] [INFO] Connect was successful
13:40:08.18 [tokio-runtime-worker-0] [opcua/client/src/comms/tcp_transport.rs:opcua_client::comms::tcp_transport] [INFO] WriteState has dropped
13:40:08.26 [tokio-runtime-worker-0] [opcua/client/src/comms/tcp_transport.rs:opcua_client::comms::tcp_transport] [INFO] ReadState has dropped
13:40:08.27 [tokio-runtime-worker-0] [opcua/client/src/comms/tcp_transport.rs:opcua_client::comms::tcp_transport] [INFO] Timer for finished is finished
13:40:08.28 [main] [opcua/client/src/session.rs:opcua_client::session] [INFO] Session has dropped
13:40:08.28 [main] [opcua/client/src/comms/tcp_transport.rs:opcua_client::comms::tcp_transport] [INFO] TcpTransport has dropped
13:40:08.28 [main] [opcua/client/src/client.rs:opcua_client::client] [INFO] Server has these endpoints:
13:40:08.28 [main] [opcua/client/src/client.rs:opcua_client::client] [INFO]   opc.tcp://10.1.1.1:1234/DataAccessServer - Basic256 / SignAndEncrypt
13:40:08.28 [tokio-runtime-worker-0] [opcua/client/src/subscription_timer.rs:opcua_client::subscription_timer] [INFO] Timer receiver has terminated
13:40:08.28 [tokio-runtime-worker-0] [opcua/client/src/session_state.rs:opcua_client::session_state] [INFO] SessionState has dropped
13:40:08.28 [main] [opcua/client/src/client.rs:opcua_client::client] [INFO] Creating a session for endpoint opc.tcp://10.1.1.1:1234/DataAccessServer, Basic256 / SignAndEncrypt
13:40:08.28 [main] [opcua/client/src/session.rs:opcua_client::session] [INFO] Connect
13:40:08.28 [main] [opcua/client/src/session.rs:opcua_client::session] [INFO] Security policy = Basic256
13:40:08.28 [main] [opcua/client/src/session.rs:opcua_client::session] [INFO] Security mode = SignAndEncrypt
13:40:08.48 [main] [opcua/client/src/session_state.rs:opcua_client::session_state] [INFO] Making secure channel request
13:40:08.48 [main] [opcua/client/src/session_state.rs:opcua_client::session_state] [INFO] security_mode = SignAndEncrypt
13:40:08.48 [main] [opcua/client/src/session_state.rs:opcua_client::session_state] [INFO] security_policy = Basic256

13:40:18.51 [main] [opcua/client/src/session_state.rs:opcua_client::session_state] [INFO] Timeout waiting for response from server
13:40:18.51 [main] [opcua/client/src/message_queue.rs:opcua_client::message_queue] [INFO] Request 1 has timed out and any response will be ignored
13:40:18.51 [main] [opcua/client/src/session.rs:opcua_client::session] [WARN] session:2 Connect was unsuccessful, error = BadTimeout, retries = 1
13:40:18.51 [main] [opcua/client/src/session_retry.rs:opcua_client::session_retry] [INFO] Retry retriggered by policy
13:40:18.51 [main] [opcua/client/src/session.rs:opcua_client::session] [INFO] Retrying to connect to server...
13:40:18.51 [main] [opcua/client/src/session.rs:opcua_client::session] [INFO] Connect
13:40:18.51 [main] [opcua/client/src/session.rs:opcua_client::session] [INFO] Security policy = Basic256
13:40:18.51 [main] [opcua/client/src/session.rs:opcua_client::session] [INFO] Security mode = SignAndEncrypt
thread 'main' panicked at 'Should not try to connect when already connected', opcua/client/src/comms/tcp_transport.rs:180:13
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace.
13:40:18.51 [main] [opcua/client/src/session.rs:opcua_client::session] [INFO] Session has dropped
thread 'tokio-runtime-worker-1' panicked at 'called `Option::unwrap()` on a `None` value', /rustc/5e1a799842ba6ed4a57e91f7ab9435947482f7d8/src/libcore/macros/mod.rs:15:40
13:40:18.51 [tokio-runtime-worker-1] [opcua/client/src/comms/tcp_transport.rs:opcua_client::comms::tcp_transport] [INFO] WriteState has dropped
```

locka99, 2020-05-27:
Can you try with RUST_BACKTRACE set to 1?

The first thing to try when crypto is involved is to ensure that the client and server both trust each other. So without knowing what your server is, check if it has rejected the client's cert and if you need to move a cert from a rejected folder into a trusted one. Since you have trust_server_certs set to true you do not need to do this on the client side.

dgl2, 2020-06-11:
Thank you for your advice.
The certificate was appearing neither in the "trusted" nor the "rejected" folder, so something must had gone wrong before that on the server. The certificate we were using was provided by our partner but there must some incompatibility as changing the certificate solved the issue. Though we still do not know which part was faulty.

I'm closing the issue as we cannot say if the problem was related to your library or our partner server implementation.

Thanks again for your help.

closed 2020-06-11 by dgl2