locka99/opcua #55: Publish Request being sent every publish interval rather than after receiving response

pulzzedavid, 2020-05-25:
I may be missing some configuration on my end. If that is the case, some guidance would be much appreciated.
The issue happens as publish request is being sent from client at subscribed publish interval rather than after receiving a response. This results in server complaining about too many publish requests.

        ServiceFault
            ResponseHeader: ResponseHeader
                Timestamp: May 24, 2020 18:47:49.946309200 PDT
                RequestHandle: 156
                ServiceResult: 0x80780000 [BadTooManyPublishRequests]
                ServiceDiagnostics: DiagnosticInfo
                StringTable: Array of String
                AdditionalHeader: ExtensionObject

locka99, 2020-05-26:
Are you using the library as a client, or the server or both? The client's behaviour is basically to throw as many publish requests at the server as there are subscriptions and to throttle if it receives an error about too many publish requests, resuming when it gets a publish response.

It should be harmless to receive the error but if it isn't let me know.

pulzzedavid, 2020-05-27:
Thanks for the clarification. So, it is by design. We are using the library as a client and server.
Will close and reopen this if there is an issue with the design.

closed 2020-05-27 by pulzzedavid