locka99/opcua #56: Failing to connect to server

TotalKrill, 2020-05-27:
Hello!

I am trying to evaluate this library, I started by trying the python client, which connects without much hassle. I was just trying out the samples found here: https://github.com/FreeOpcUa/python-opcua/blob/master/examples/client-minimal.py

I cannot however get this client to connect to the same server I am trying against. I am using this code:

locka99, 2020-05-27:
Can you try the thing again with RUST_BACKTRACE set to 1 to see exactly where in the stack you were getting the error? Setting RUST_OPCUA_LOG to debug can also yield useful info, often identifying the problem connecting with a server.

In general, the error you are seeing is in a function called send_message() which attempts to send messages from one thread (typically the main thread) to the OPC UA thread. If the OPC UA thread has terminated then you'll get a panic.

nathan-brainkey, 2021-08-09:
@TotalKrill, did you manage to figure this out? I am having the same issue.

nathan-i-strong, 2021-08-09:
@TotalKrill , I am having the same issue. did you manage to figure it out?

@locka99 , I also had issues when replicating the simple client. I got the error message that @TotalKrill is reporting. Do you have any suggestions on how to fix it? Super appreciated!

locka99, 2021-08-10:
You'll have to describe a little more what server you are trying to connect to and the endpoint - url, security policy, security mode, token. e.g. if there is crypto then the sample above doesn't create a keypair (it says create_sample_keypair(false)) so it would fail if something else didn't create a cert / private key.

sysarcher, 2021-08-11:
Potentially related to: #121  ??

I think the errors look identical... @nathan-i-strong can you try adding the timeout?

milgner, 2021-09-23:
Also just stumbled on the same error message when trying to read a node.
The connection gets established just fine, but when trying to invoke `session.read` to read a node, the error above occurs.

TCP connection, security policy `None`, message security mode `None` (to keep things as simple as possible).
The goal is to browse the server and in order to get the list of available namespaces, I'm trying to do it like this:

```rust
const SERVER_NAMESPACES_NODE_ID: &str = "i=2255";

let nodesToRead = [
    ReadValueId {
        node_id: NodeId::new(0, SERVER_NAMESPACES_NODE_ID),
        attribute_id: AttributeId::Value as u32,
        index_range: UAString::null(),
        data_encoding: QualifiedName::null()
    }
];
let mut writeSession = session.write().unwrap(); // the original session is the Arc<RwLock<Session>>
let readData = writeSession.read(&nodesToRead) 
```

Full backtrace:

```
thread '<unnamed>' panicked at 'called `Option::unwrap()` on a `None` value', /home/marcus/.cargo/registry/src/github.com-1ecc6299db9ec823/opcua-client-0.8.1/src/message_queue.rs:53:48
stack backtrace:
   0: rust_begin_unwind
             at /rustc/c8dfcfe046a7680554bf4eb612bad840e7631c4b/library/std/src/panicking.rs:515:5
   1: core::panicking::panic_fmt
             at /rustc/c8dfcfe046a7680554bf4eb612bad840e7631c4b/library/core/src/panicking.rs:92:14
   2: core::panicking::panic
             at /rustc/c8dfcfe046a7680554bf4eb612bad840e7631c4b/library/core/src/panicking.rs:50:5
   3: core::option::Option<T>::unwrap
             at /rustc/c8dfcfe046a7680554bf4eb612bad840e7631c4b/library/core/src/option.rs:722:21
   4: opcua_client::message_queue::MessageQueue::send_message
             at /home/marcus/.cargo/registry/src/github.com-1ecc6299db9ec823/opcua-client-0.8.1/src/message_queue.rs:53:27
   5: opcua_client::message_queue::MessageQueue::add_request
             at /home/marcus/.cargo/registry/src/github.com-1ecc6299db9ec823/opcua-client-0.8.1/src/message_queue.rs:63:9
   6: opcua_client::session_state::SessionState::add_request
             at /home/marcus/.cargo/registry/src/github.com-1ecc6299db9ec823/opcua-client-0.8.1/src/session_state.rs:331:9
   7: opcua_client::session_state::SessionState::async_send_request
             at /home/marcus/.cargo/registry/src/github.com-1ecc6299db9ec823/opcua-client-0.8.1/src/session_state.rs:279:9
   8: opcua_client::session_state::SessionState::send_request
             at /home/marcus/.cargo/registry/src/github.com-1ecc6299db9ec823/opcua-client-0.8.1/src/session_state.rs:244:30
   9: opcua_client::session::Session::send_request
             at /home/marcus/.cargo/registry/src/github.com-1ecc6299db9ec823/opcua-client-0.8.1/src/session.rs:2379:9
  10: opcua_client::session::Session::read
             at /home/marcus/.cargo/registry/src/github.com-1ecc6299db9ec823/opcua-client-0.8.1/src/session.rs:1444:28
```

schroeder-, 2021-09-23:
I think you want this for the nodeid:
`const SERVER_NAMESPACES_NODE_ID: i32 = 2255;`
or even better use this:
`const SERVER_NAMESPACES_NODE_ID: i32 = VariableId::Server_NamespaceArray as i32;`

milgner, 2021-09-23:
> VariableId::Server_NamespaceArray as i32

@schroeder- Much appreciated! Hadn't see that definition, yet. Unfortunately it doesn't change anything about the error.

schroeder-, 2021-09-23:
I used the following code with version "0.8.1" against opc.tcp://opcua.umati.app:4840 and it didn't encounter a problem:
```
fn test_read(session: Arc<RwLock<Session>>) -> Result<(), StatusCode>{
    const SERVER_NAMESPACES_NODE_ID: i32 = VariableId::Server_NamespaceArray as i32;
    let nodes_to_read = [
        NodeId::new(0, SERVER_NAMESPACES_NODE_ID).into()
    ];
    let mut write_session = session.write().unwrap(); 
    let read_data = write_session.read(&nodes_to_read);
    match read_data{
        Ok(vv) => {
            for v in vv{
                println!("{:?}", v)
            }
        },
        Err(err) => println!("Error {}", err),
    }
    Ok(())
}
```

schroeder-, 2021-09-23:
After looking in the stacktrace and checking the function MessageQueue::send_message I think it's not the posted code. The problem is in creating the connection. Can you post your code where you create the session?

milgner, 2021-09-24:
> After looking in the stacktrace and checking the function MessageQueue::send_message I think it's not the posted code. The problem is in creating the connection. Can you post your code where you create the session?

Once more thank you for the quick reply! It sounds plausible that this might be the case - after upgrading the library from 0.6 to 0.8, establishing the connection takes a couple of seconds to go through which is quite unexpected as the server responds quickly with every other client.

The session is established like this:

```
// running in the context of a struct that encapsulates the client and session
//    wrapped: Option<opcua_client::prelude::Client>,
//    session: Option<std::sync::Arc<std::sync::RwLock<opcua_client::prelude::Session>>>,

let server_endpoint = opcua_client::prelude::Client::find_server_endpoint(&endpoints,
                                                                          self.uri.as_str(),
                                                                          self.security_policy,
                                                                          self.message_security_mode);
if server_endpoint.is_none() {
    println!("Failed to find server endpoint");
    return Some(std::io::Error::from(ErrorKind::NotFound));
}
let user_identity_token = match self.username.len() {
    0 => opcua_client::prelude::IdentityToken::Anonymous,
    _ => IdentityToken::UserName(self.username.clone(), self.password.clone())
};
// after upgrading from 0.6 to 0.8 this takes multiple seconds to connect
let session = self.wrapped.as_mut().unwrap()
    .connect_to_endpoint(server_endpoint.unwrap(), user_identity_token);

match session {
    Ok(session) => {
        self.session = Some(session); // maybe the move here is a problem?
        None // no error
    },
    Err(code) => {
        println!("Failed to connect session to server endpoint: {}", code.name());
        Some(std::io::Error::from(ErrorKind::ConnectionRefused))
    }
}
```

milgner, 2021-09-24:
Update: tried this variation and got `BadNotConnected` error after invoking `connect()`

```rust
match session {
    Ok(session) => {
        println!("Got session");
        println!("Session is connected? {}", session.read().unwrap().is_connected());
        let connectResult = session.write().unwrap().connect();
        match connectResult {
            Ok(_) => {
                self.session = Some(session); // maybe the Move here is a problem?
                None
            }
            Err(error) => {
                println!("Failed to connect session: {}", error);
                Some(std::io::Error::from(ErrorKind::ConnectionRefused))
            }
        }
    },
    Err(code) => {
        println!("Failed to connect session to server endpoint: {}", code.name());
        Some(std::io::Error::from(ErrorKind::ConnectionRefused))
    }
}
```

milgner, 2021-09-24:
Another update: I saw the `opcua_console_logging` crate in some of the samples and activated it. This is the output:

```
2021-09-24 08:20:30.706 - ERROR - opcua_core::comms::secure_channel - Cannot read our own certificate and private key. Check paths. Crypto won't work
2021-09-24 08:20:31.007 - ERROR - opcua_core::comms::secure_channel - Cannot read our own certificate and private key. Check paths. Crypto won't work
2021-09-24 08:20:31.036 - ERROR - opcua_client::comms::tcp_transport - Invalid address 8358eb103f8a:4840, cannot be parsed Custom { kind: Uncategorized, error: "failed to lookup address information: Name or service not known" }
2021-09-24 08:20:31.048 - ERROR - opcua_client::comms::tcp_transport - Invalid address 8358eb103f8a:4840, cannot be parsed Custom { kind: Uncategorized, error: "failed to lookup address information: Name or service not known" }
2021-09-24 08:20:41.045 - ERROR - opcua_client::comms::tcp_transport - Invalid address 8358eb103f8a:4840, cannot be parsed Custom { kind: Uncategorized, error: "failed to lookup address information: Name or service not known" }
2021-09-24 08:20:41.045 - ERROR - opcua_client::session - session:2 Session has given up trying to connect to the server after 3 retries
2021-09-24 08:20:41.045 - ERROR - opcua_client::client - Got an error while creating the default session - BadNotConnected
Got session
Session is connected? false
2021-09-24 08:20:41.052 - ERROR - opcua_client::comms::tcp_transport - Invalid address 8358eb103f8a:4840, cannot be parsed Custom { kind: Uncategorized, error: "failed to lookup address information: Name or service not known" }
2021-09-24 08:20:41.052 - ERROR - opcua_client::session - session:2 Session has given up trying to connect to the server after 4 retries
Failed to connect session: BadNotConnected
```

So it seems to be a problem with the server which returns weird endpoint information in its reflection. Nonetheless, I'd have expected the library to not return a valid session if `connect_to_endpoint` fails to connect.