locka99/opcua #58: User/Password authentification fails on SecurityPolicy#None

execuc, 2020-07-09:
Hello,

I use your library, which is very good and reliable, as a client to connect to a opcua server. 
Server endpoint is described as :

```yaml
EndpointDescription
    EndpointUrl: opc.tcp://a.b.c.d:12345
    Server: ApplicationDescription
[...]
    MessageSecurityMode: None (0x00000001)
    SecurityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#None
    UserIdentityTokens: Array of UserTokenPolicy
        ArraySize: 1
        [0]: UserTokenPolicy
            PolicyId: UserName
            UserTokenType: UserName (0x00000001)
            IssuedTokenType: [OpcUa Null String]
            IssuerEndpointUrl: [OpcUa Null String]
            SecurityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
    TransportProfileUri: http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary
    SecurityLevel: 16
```

So there is no signature or encryption for message but password token has to be sent in RsaPkcs because of token Basic128Rsa15. But authentification is rejected by the server with a BadIdentityTokenInvalid error at ActivateSessionRequest state.

Token policy
------------

My first problem is I have to force the use of Basic128Rsa15 policy in `make_user_name_identity_token` function because the library seems to take the endpoint policy (None) instead of token policy resulting a clear password.

Is this a configuration problem on my side ?

Token content
-------------

My second problem is that authentication fails with the same error. In this case, the content of the token is incomplete before it is encrypted in `legacy_password_encrypt`. I noticed that the server_nonce is empty. In fact it is not set in `set_remote_nonce_from_byte_string` function because of endpoint None policy.

Is it a problem of endpoint/token mode of the server which is not possible in the opcua specifications or it is not managed by the library.

Thank you for your work

laumann, 2021-06-15:
Hi @execuc I want to jump in on this issue, as I think we've been grappling with the same problem. I opened #111 (by mistake), but I wanted to follow up here.

Our setup is basically identical to the one you describe: Endpoint security policy = None, message security mode = None, but the user token policy = Basic128rsa15. We've worked from the 0.8 tag exclusively.

On the token policy: We figure that the logic around determining the security policy is wrong, but we could be mistaken. It doesn't appear to be a configuration issue.

On the token content: It turns out that the server nonce isn't stored in the secure_channel, because it decides that because the security policy and mode both are None, then we shouldn't keep the nonce. But we _do_ need the nonce for encrypting the password.

See jhttps://github.com/omnioiot/opcua/pull/1 for the implementation that we're currently using.

laumann, 2021-06-17:
@locka99 I'd be very interested in opening a PR to address this issue - do you have any input on how it should work?

execuc, 2021-06-17:
Hi @laumann,
I don't know what I have changed locally to make it work. In all honesty it's been a year since I last did opcua so I'm a bit lost. In addition, I do not currently have access to my code but I have to get back to it in 2 weeks :)

laumann, 2021-06-17:
> I don't know what I have changed locally to make it work. In all honesty it's been a year since I last did opcua so I'm a bit lost. In addition, I do not currently have access to my code but I have to get back to it in 2 weeks :)

I know the feeling :-) Let me know if you'd like to collaborate on the solution. The PR that I linked previously I've only tested by mangling the simple-client sample to use username/password authentication. But we're testing this change in an integrated solution, so we'll see if we run into any problems.

laumann, 2021-06-17:
@execuc The way I understand it is that there are two levels of "security policy". For each endpoint we have the pair (security policy, message security policy), but there's an additional security policy for each defined user identity token. This allows for the situation that for the opened connection there's security policy None and message security policy None, _but_ authentication to open the connection can be security policy Basic128Rsa15.

It looks to me like the two levels of security policy are conflated in the code that (1) determines the security policy used for authentication and (2) determines whether we should keep the server's provided nonce.

I'm not sure I'm right though, which is why I would like to hear from @locka99 :smile:

schroeder-, 2021-06-17:
That's correct servers can request to send the authentication token (in this case password), to be send via a server selected security policy. For example in a commercial C++ Server SDK you can set this explicit.

locka99, 2021-06-17:
I'm happy to accept a PR if you have a good solution.

Summary of current behaviour from the client side when sending identity token to server is:

1. Security policy == None, use plaintext password
2. Security policy != None, encrypted legacy password with encryption algorithm, nonce, padding derived from security policy

My feeling is if you need to send a pwd you should be using encryption.

I've never been 100% clear on the rules for nonce when the security mode is None.  The OPC UA part 2 doc defines the SecureChannelNonceLength for encryption modes, usually specifying 32 bytes or 16 bytes for Basic128Rsa15 but doesn't say for None, nor how long the nonce should be or if it should be filled with random values or not.

Part 6 also says - "The OpenSecureChannel Messages are not signed or encrypted if the SecurityMode is None. The Nonces are ignored and should be set to null."

But the OPC UA compliance tests expect ALL of the following to succeed for a None connection:

1. Provide certificates & nonce, expect channel to open
2. Provide nonce, no certs, expect channel to open
3. Do not provide cert or nonce, expect channel to open
4. Provide invalid cert, expect channel to open

So while a server might succeed for any of these scenarios it might not necessarily work well with passwords.

Concerning password encryption, OPC UA part 4 & 7 describe UserNameIdentityToken, UserTokenPolicy and the encryption algorithm behaviours. In particular 7.36.2 in part 4. 

At present the code that makes the user identity token is in the function `make_user_name_identity_token()` in crypto/src/user_identity.rs. 

I do not have support for the EncryptedSecret format which is super complicated and I haven't even looked at how I would implement it.

laumann, 2021-06-17:
@schroeder- Thanks, good to know!

Just for some background information, we are connecting to a Kepware instance through it's OPC-UA server interface and get this when connecting:
```
14/06/2021 15.25.53.735 [AnonymousClient] CreateSessionResponse
	0000000000: Event started
	0000000000:    Response Header: 
	0000000000:       timestamp (UTC): 2021-06-14T13:25:53.735 
	0000000000:       requestHandle: 2 
	0000000000:       serviceResult: 0x00000000 (Good) 
	0000000000:    Parameters: 
	0000000000:       sessionId: ns=1;g={2b25e62e-31f0-43ea-a7e5-e1907d337643} 
	0000000000:       authenticationToken: i=1660570039 
	0000000000:       revisedSessionTimeout: 15000 
	0000000000:       serverNonce: b4697bc16c66cd3182eee8d79d2902f8003b747a45c69dc2002229ae7783df95 
	0000000000:       serverCertificate (thumbprint): b00a0b9e8d685beb13bfbccad191d00e0772a3db 
	0000000000:       serverEndpointDescriptions []: Size: 6 
--- [snip] ---
	0000000000:          serverEndpointDescriptions [ 5 ]: 
	0000000000:             endpointUrl: opc.tcp://192.168.10.11:49320 
	0000000000:             serverApplicationDescription: 
	0000000000:                applicationUri: urn:Win10Edge:Kepware.KEPServerEX.V6:UA%20Server 
	0000000000:                productUri: urn:Win10Edge:Kepware.KEPServerEX.V6:UA%20Server 
	0000000000:                applicationName: en|KEPServerEX/UA@Win10Dev.local 
	0000000000:                applicationType: Server 
	0000000000:                gatewayServerUri: [empty] 
	0000000000:                discoveryProfileUri: [empty] 
	0000000000:                discoveryUrls []: Size: 2 
	0000000000:                   discoveryUrls: [ 0 ]: opc.tcp://Win10Dev.local:49320 
	0000000000:                   discoveryUrls: [ 1 ]: opc.tcp://192.168.10.11:49320 
	0000000000:             serverCertificate (thumbprint): b00a0b9e8d685beb13bfbccad191d00e0772a3db 
	0000000000:             messageSecurityMode: None <-- message security mode
	0000000000:             securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#None <-- security policy
	0000000000:             userIdentityTokens []: Size: 2 
	0000000000:                userIdentityTokens [ 0 ]: 
	0000000000:                   policyId: UserName 
	0000000000:                   tokenType: 1 
	0000000000:                   issuedTokenType: [empty] 
	0000000000:                   isssuerEndpointUrl: [empty] 
	0000000000:                   securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15 <-- security policy for user identity token
	0000000000:                userIdentityTokens [ 1 ]: 
	0000000000:                   policyId: Anonymous 
	0000000000:                   tokenType: 0 
	0000000000:                   issuedTokenType: [empty] 
	0000000000:                   isssuerEndpointUrl: [empty] 
	0000000000:                   securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#None 
--- [snip] ---
	0000000000: Event complete
```

With the changes I made in omnioiot/opcua#1 I get the following (successful) connection:

```
14/06/2021 15.25.53.841 [Rust OPCUA Client] ActivateSessionRequest
	0000000000: Event started
	0000000000:    Request Header: 
	0000000000:       authenticationToken: i=1660570039 
	0000000000:       timestamp (UTC): 2021-06-14T13:25:54.877 
	0000000000:       requestHandle: 3 
	0000000000:       returnDiagnostics: 0 
	0000000000:       auditEntryId: [empty] 
	0000000000:       timeoutHint(ms): 10000 
	0000000000:    Parameters: 
	0000000000:       clientSignature: 
	0000000000:          algorithm: [empty] 
	0000000000:          signature: [empty] 
	0000000000:       localeIds []: Size: 0 
	0000000000:       userIdentityToken: 
	0000000000:          parameterTypeId: 324 
	0000000000:          UserNameIdentityToken: 
	0000000000:             policyId: UserName 
	0000000000:             userName: ****** 
	0000000000:             password: 75ad995b4034437ebda198618c89bbf0e9da12e4cfa446fec11f556d006ec98ead83c80b42cbaaad1372561573707bca34244512a42670e979e871741475447186e9c518c387842a9dd69f3c6ffddad3da33525cd99987a7ccb86b7a26912ca08b801d69508c75709a45c5296a5efc3438f53468b6db531d3e0290716baf6b39edc9fca67d115a1e17ea196e3aeb4bd472ddaf81f17f8bc7f63cd5b276d8fc3d194591b84c71062fc576a20c3709e2cae2e3db503660d5ae34b83e77f9b1a55c66bf004c028f34d3d77422dd70bf6484f5ced7c72d23f3eda2c6cf0550e865fa840fae80aad63c78d38e1b1e01a3b13b20b72bdc36a489dbbc343ce218830600 
	0000000000:             encryptionAlgorithm: http://www.w3.org/2001/04/xmlenc#rsa-1_5 
	0000000000:       userTokenSignature: 
	0000000000:          algorithm: [empty] 
	0000000000:          signature: [empty] 
	0000000000: Event complete
```

From what I gather, we can open a connection with security policy/message security mode as None/None, but if we authenticate with a username and password, we must use security policy Basic128Rsa15 when sending the password.

@locka99 Thanks for the info, I'll go look at  the OPC-UA spec and try to figure out how it should work.

laumann, 2021-11-17:
@execuc I think this should be fixed now. @locka99 you can probably close this issue.

locka99, 2021-11-17:
Okay thanks I'll close tgis

closed 2021-11-17 by locka99