locka99/opcua #59: A response with request handle 0 doesn't belong to any request and will be ignored

svanharmelen, 2020-09-23:
I get a lot of these errors when using the OPC UA client:

```2020-09-23 08:51:26.108 - ERROR - opcua_client::message_queue - A response with request handle 0 doesn't belong to any request and will be ignored, inflight requests = {(7, true), (11, true), (10, true), (12, true), (13, true)}
2020-09-23 08:51:27.101 - ERROR - opcua_client::message_queue - A response with request handle 0 doesn't belong to any request and will be ignored, inflight requests = {(7, true), (11, true), (10, true), (12, true), (13, true), (14, true)}
2020-09-23 08:51:28.109 - ERROR - opcua_client::message_queue - A response with request handle 0 doesn't belong to any request and will be ignored, inflight requests = {(7, true), (11, true), (10, true), (12, true), (13, true), (14, true), (15, true)}
2020-09-23 08:51:29.112 - ERROR - opcua_client::message_queue - A response with request handle 0 doesn't belong to any request and will be ignored, inflight requests = {(14, true), (10, true), (13, true), (16, true), (7, true), (11, true), (12, true), (15, true)}
2020-09-23 08:51:30.101 - ERROR - opcua_client::message_queue - A response with request handle 0 doesn't belong to any request and will be ignored, inflight requests = {(14, true), (10, true), (13, true), (16, true), (17, true), (7, true), (11, true), (12, true), (15, true)}
```

I'm connecting to a 3rd party OPC UA server (well known brand) and do also get the expected item updates. Anything I can or should do to mitigate or investigate this?

I have a bit of a hard time understanding what this means or what causes these messages... Thanks!

svanharmelen, 2020-09-23:
Ah, I found that every message like this corresponds to a `ServiceFaultResponse`:

[image]

Funny thing is that the response does have the correct handle, but they seem to have been send out of order. Anything I can/should do to fix this?

svanharmelen, 2020-09-23:
Found the problem, will make a PR to solve it 👍🏻

locka99, 2020-09-23:
Hi Sander,

The request handle is supplied by the client, e.g opcua-client, and is
usually an incrementing number in each request that allows it to match up
when it receives the corresponding response.  I think my code always
returns a number of 1 or higher that increments with each request. If the
client isn't setting the handle in the request or the server isn't
returning the same number in the response then there will be trouble
matching the request / response to the other.

When you say you found the issue, was it it my code or somewhere else? My
client code initialises the handle to 1 and each request calls to
make_request_header() that should return a header with the handle
incremented by 1, but it's possible I've overlooked something.

Thanks

Adam

locka99, 2020-09-23:
Followup - It's possible that there is also an ordering issue, that a
response doesn't arrive in the order that the request was sent. So possibly
the client code hasn't been written to handle these interleaved responses
properly. If you find that to be the case, let me know and I can look at
the situation.

Adam

svanharmelen, 2020-09-23:
The problem I seemed to have encountered was related to the `ServiceFault` not being in the response list. So it seems to be fixed by your change and the additional tweak in PR #60 

Thanks for the quick response!

locka99, 2020-09-24:
Thanks for the pr, I've just accepted it there

closed 2020-09-24 by locka99

svanharmelen, 2020-09-24:
Thanks @locka99!

PR #60 (2020-09-24): Add ServiceFault to`response_header`
In addition I sorted al the big lists to make it easier to find anything in them. Let me know if you want that part reverted…

The only logical change is on lines 222 and 267.

Fixes #59