locka99/opcua #61: Is version 0.7 a "stable" version that can be used (relatively) safely?

svanharmelen, 2020-09-23:
And how about (the work in progress) version 0.8? Would like to know your take on if these (or at least version 0.7) could be used in a production like setting? And also if it's save to use the client within a Tokio code base.

It looks like the `async_run` function is made to spawn a new dedicated OS thread and then does all the scheduling in that single thread. Is that correct? If so I guess it can be safely used as it will actually have it's own runtime in its own thread. But good to be sure and/or if there are any other caveats to take into account.

Thanks! And also thanks for this extensive crate!

locka99, 2020-09-24:
OPC UA is a very complicated standard that runs at various profile levels nano, micro, embedded & standard. Up to and including 0.7 I was mostly getting the implementation into a state where it was close enough to the embedded spec that 3rd party implementations could talk to it in a typical session. But even so I was missing a few service requests and there were plenty of edge cases where I wasn't handling bad requests, or returning the right code.

So for 0.8 I've gone over the server side trying to get the implementation of every service request as well as core address space vars to comply to the standard to the letter, as close as I can, up to at least micro compliance with embedded where practible. That's why it's dragging on a bit but you can refer to the CHANGELOG to see what I'm doing.

The code in the server is asynchronous through tokio 0.1. Tokio means in theory that it will scale better than dedicating one thread per client. In the client I run the session's processing loop on a separate thread so that your main thread, the thing calling functions on OPC UA is not blocked by async publish requests or other house keeping. At the moment the API is mostly synchronous  so you call and wait, but underneath it is not synchronous so in the future I might open that up.

I also intend in 0.9 to move to tokio 0.2 which uses the std::Future.

closed 2020-09-24 by locka99

svanharmelen, 2020-09-24:
Thanks for the details @locka99. Yet one thing still isn't super clear to me... Say that I have an application that is already using Tokio 0.2 like this:

```
#[tokio::main]
pub async fn main() -> Result<()> {
    ...
}
```

In this case the application already creates a multi-threaded Tokio runtime and it would potentially cause issues if another runtime or executor is created/started within the existing one. Now I only intent to use the client, but I want to make sure that it doesn't try to nest a runtime or executor.

I looked through the code, but I fail to really understand how/where you create your runtime. Can you point me in the right direction so I can get a better feeling of how and where you use Tokio for a client? Thanks!

svanharmelen, 2020-09-24:
Ah... I'm not familiar with Tokio 0.1, but after reading up a bit it looks like `tokio::run` is what you use to start a runtime with Tokio 0.1.

Reading through the code searching for all uses of `tokio::run`, it looks like it is only used inside of newly created dedicated OS threads (which means that the created runtime/executor will not be nested in the one I already started) except for this one here: https://github.com/locka99/opcua/blob/master/server/src/server.rs#L247

But that is the main server loop which I will not be using when I only create and use a client, right? So if these conclusions are correct, using a client within a Tokio 0.2 app should be fine but using the server could cause some issues...

locka99, 2020-09-24:
I don't think multiple tokios will interfere with each other because name mangling would make them independent of one another.

I haven't looked at 0.2 in too much detail but the old tokio basically has a tokio::run() command that takes a top level task that potentially spawns other tasks and blocks until all the tasks are done. So calling run() blocks the thread.

My impl Server::run() calls tokio::run and so it also blocks the thread. For servers that don't want that, they can spawn on a separate thread. My top level task sets up a socket listener task which in turn spawn additional tasks to handle each client connection. Everything is async so that tokio::run is cycles through the open tasks until they signal they're done. I don't explicitly set how many worker threads tokio uses for work, so it might spawn its own but this async model is more scalable than if I spawned a thread per client.

Client side is a little different but the same principle applies. You want a client to be able to call functions that map to OPC UA requests but the client's session also needs to do housekeeping tasks like checking the socket is alive and sending publish events. These are all tokio tasks. Some clients might be happy to invoke Session::run() and block forever (e.g. subscription listeners while others might have other things they want to call and prefer to call Session::run_async() which spawns another thread that does the blocking while they get on with their own thing.

svanharmelen, 2020-09-24:
Yep, sounds good and logical 👍🏻 In my case the whole application is based on async/await logic, so I have one multi-threaded Tokio runtime and only call `tokio::spawn` whenever I need to spawn an async task.

As you create the runtimes in their own threads (and have the `async_run` method for the client), I don't see any issues with that for my code.

I started to implement some parts in our application, will let you know if I bump into anything unexpected. Thanks again!