locka99/opcua #65: How can I get the raw value out of a Variant?

svanharmelen, 2020-10-02:
I want to serialize the value of a changes monitored item as part of another struct that has a field of type `serde_json::Value` so that I can put all kinds of values in there. But if I give it a received value it serializes to something like this:
```
{"my_value":{"Double":7247.500215355554}}
```

While I would like to get something like this:
```
{"my_value": 7247.500215355554}
```

I tried using `value.0` but (at least in my code/setup) the compiler doesn't allow that. So any idea how to achieve this? Thanks!

locka99, 2020-10-02:
I think you'd have to write your own custom code because the value of a variable in OPC UA can be one of 20 or so types (and arrays) so there is a [Variant](https://github.com/locka99/opcua/blob/master/types/src/variant.rs) enum type in my crate for that purpose. Variant implements serialize and serde is obviously choosing to represent it in JSON in the way you describe.

If you absolutely know a value is a double, then copy the value out of the DataValue.value as a Double into your own type that implements Serialize and it should appear as you like.

Also, you might want to look at the [web-client](https://github.com/locka99/opcua/blob/master/samples/web-client/src/main.rs) sample because that turns values into JSON via a DataChangeEvent and has Javascript to extract the value.

closed 2020-10-02 by locka99

svanharmelen, 2020-10-03:
Thanks for the info! I'll have a look at the web-client 👍🏻