locka99/opcua #72: Aes256-Sha256-RsaPss uses RSA-OAEP-SHA2-256 for asymmetric encryption that requires 66 bytes for padding

ColinFinck, 2020-10-27:
The current implementation always assumes RSA-OAEP-SHA1 when encrypting with OAEP padding and therefore only leaves a padding margin of 42 bytes:

https://github.com/locka99/opcua/blob/5103770a9fd9ac597e5dd2ae880bf4b08b9ebf88/crypto/src/pkey.rs#L73

However, when using a SHA256 hash for OAEP, we need 66 bytes padding margin.

References:
- https://crypto.stackexchange.com/a/66374
- https://security.stackexchange.com/a/112032
- https://opensource.apple.com/source/Security/Security-58286.41.2/keychain/SecKey.h

locka99, 2020-10-27:
Thanks for the background info. I have the mode down as something to do for 0.9 in the changelog. I had most of the code path for it enabled in 0.8 but it wasn't working (probably padding as you say) so I marked it as unsupported.

locka99, 2020-10-28:
This proves to be a bit more complicated than I was hoping for. Basically the OpenSSL function RSA_private_decrypt / encrypt don't have knowledge of OAEP padding with SHA256 so a workaround would have to be performed, one that works through the openssl crate.

Basically calling a bunch of raw OpenSSL functions to perform the same task.

https://stackoverflow.com/questions/17784022/how-to-encrypt-data-using-rsa-with-sha-256-as-hash-function-and-mgf1-as-mask-ge

locka99, 2020-10-31:
The mode has been implemented. I had to write a bunch of primitive calls over OpenSSL because unlike regular OAEP-SHA1 there is no in-built function for just doing it.

closed 2020-10-31 by locka99