locka99/opcua #79: how to send `TranslateBrowsePathsToNodeIDsRequest` from client?

nkbai, 2020-12-02:
it seems that the client module miss this interface.

locka99, 2020-12-02:
I can implement it on the master thread if you like, shouldn't be hard to reproduce

locka99, 2020-12-02:
I checked in 3cd371cb on master for this. It is a wrapper around the request / response so it should not be complex.

nkbai, 2020-12-03:
@locka99  but session's `send_request` is private function.

locka99, 2020-12-03:
Can you clarify what the issue is with the implementation? Normally you would open a session and then call this function which would send the TranslateBrowsePathsToNodeIdsRequest  and wait for the TranslateBrowsePathsToNodeIdsResponse.  The send_request is private but it is used by all the other server calls so it should work pretty much the same.

nkbai, 2020-12-03:
as a user of  opcua-client.
1.  import the dependency opcua-client in the cargo.toml
2. create a client and get a session
3. call session.send_request, but  I can't call this function because it's not public.

locka99, 2020-12-03:
The patch though adds a public translate_browse_paths_to_node_ids() that calls the request so there should be no reason to call send_request directly.

nkbai, 2020-12-03:
yes,that's ok. 

All I’m saying is add a function like this

locka99, 2020-12-08:
I think if the client ever offers a sync / async API that issues like this will become more clear. I'm thinking when the tokio 0.2 update comes in that I could move to a more promise / futures based approach where send_request could be eliminated entirely. For the time being the API is synchronous and providing there is a method corresponding to every service request it should not be necessary to expose send_request although I could certainly make it public just in case the request needs to do something with diagnostics which isn't supported by the current api.

locka99, 2022-02-20:
I'm closing since there is a solution for this in the interim

closed 2022-02-20 by locka99