locka99/opcua #80: cannot

nkbai, 2020-12-03:


nkbai, 2020-12-03:
I can connect with `prosys demo server` uaexpert, and there is no error when browsing and calling. 
but with opcua client, there are too many errors when connect with localhost prosys demo server. 

this server can download from [here](https://downloads.prosysopc.com/opc-ua-simulation-server-downloads.php)

locka99, 2020-12-03:
I installed the server and preliminary analysis is this. The Rust client is calling get_endpoints() and getting back a single endpoint "opc.tcp://xxx:53530/" that doesn't match on the "opc.tcp://xxx:53530/OPCUA/SimulationServer" URL the user interface says is the endpoint. So if you connect with the path it just fails after discovery. You might get further by removing the path in the URL but I'll leave this open until I understand where the problem lies.

locka99, 2020-12-08:
This OPC server is proving to be a pain:

1. If I use the URL returned from a matching GetEndpoints it causes the server to throw a ServiceFault response with no request_handle so my logic doesn't know what request the response is returning for. I have hacked it in to force it to use the url it shows in the UI despite it not being the endpoint returned by GetEndpoints.
2. The CreateSessionRequest fails if I don't supply a ClientNonce even for a security policy of None. I have added 32 length array of zeros to bypass this but I'm worried if I will break other servers
3. My client's HEL explicitly says MaxChunkCount 1 and send buffer size 65k but the ACK returns MaxChunkCount 0 and buffer size 8Kb
4. I discover the it sends CreateSessionResponse in 5 chunks despite my HEL saying I only support 1 chunk.

This is as far as I got with this. I think for 4) I'm going to have to look at supporting multiple chunks. The code semi supports multiple chunk encodings but I've never hooked it up for real. I think if I get past that it might actually function, but with a lot of tidy up necessary

nkbai, 2020-12-08:
for question 2:
empty Client Nonce is ok
for question 4:
I have implemented a multiple chunks support , I will send the PR later.

locka99, 2020-12-09:
i have a patch which works with prosys but I have failing tests and will have to fix them tomorrow before pushing

locka99, 2020-12-09:
I think this is fixed now on master. Basic two issues:

1. The server_endpoint being passed to server had the path removed which confused the server. Ensuring the path made it return all of its endpoints.
2. The server returning messages in chunks, fixed by your patch.
3. Nonce was empty, now I set 32 bytes of zeroes but this may change if it causes an issue with further servers

In addition I've made some changes in the code as follows:

1. simple-client sends a product uri
2. The chunker::encode() is now called with a 0 chunk count in most places, i.e. it won't send anything as chunks although the facility in the function is still there and tested by the unit tests. So at present the server won't chunk any responses, nor will the client chunk any requests. 
3. Places that used to say max_chunk_size now say max_chunk_count and the value is propagated a little further although not completely.

closed 2020-12-09 by locka99