locka99/opcua #92: imp Error for StatusCode

smorokin, 2021-02-18:
Disclaimer: I'm really new to rust and wanted to try something out with this crate. So please feel free to close this issue, if it is a bad idea / unidiomatic, etc. 

I followed the examples to set up a client and got some crashes on the unwraps. To investigate I tried to replace them with `?`-operators, but sadly a lot of the methods return `Result<T, StatusCode>` and the trait `Error` is not implemented for `StatusCode`. 

As far as I understand I can't implement a trait for something from another crate.

Would it be possible to provide an implementation for `Error` from your side? I think it could just call it's `name` and `description` functions.

locka99, 2021-02-19:
I should be able to implement the trait but let me investigate

locka99, 2021-02-19:
A quick look at Error suggests implementing source() might be tricky but I'll take a look in more detail tomorrow

locka99, 2021-02-21:
I've implemented Error on StatusCode and left the default implementation of source() that returns None.

closed 2021-02-21 by locka99

smorokin, 2021-02-22:
Great. Thank you very much for your time and the fast implementation. Returning `None` in `source` should be the richt choice as far as I understand it.

smorokin, 2021-02-23:
Sorry to bother you again @locka99, but where can I get the new version with `Error`?

locka99, 2021-02-23:
Hi you'd have to git clone the master branch to get it.

smorokin, 2021-02-23:
I have tried that, but it is not included in master.

locka99, 2021-02-23:
You're right, sorry I forgot to push it.