locka99/opcua #97: Limits With Subscriptions Server?

morphace, 2021-03-09:
When I try to set the queue size or sample interval, it seems the opcua-server is sending much lower revised values (queue size becomes 10, if I try to set 100).

The background is that I'm trying to sucbscribe to values that change every 10ms (actually I would like to go down to 1ms).

Is that supposed to work? Why is the sever limiting the queue size to 10?

I can see that in subscription.rs:

locka99, 2021-03-09:
At the moment there are constants defined in the server\src\lib.rs here:

    /// The polling interval in millis on subscriptions and monitored items. The more
    /// fine-grained this is, the more often subscriptions will be checked for changes. The minimum
    /// publish interval cannot be less than this.
    pub const SUBSCRIPTION_TIMER_RATE_MS: u64 = 100;
    /// Minimum publishing interval for subscriptions
    pub const MIN_PUBLISHING_INTERVAL: f64 = (SUBSCRIPTION_TIMER_RATE_MS as f64) / 1000.0;
    /// Minimum sampling interval on monitored items
    pub const MIN_SAMPLING_INTERVAL: f64 = (SUBSCRIPTION_TIMER_RATE_MS as f64) / 1000.0;
    /// Maximum data change queue allowed by clients on monitored items
    pub const MAX_DATA_CHANGE_QUEUE_SIZE: usize = 10;

This is why the settings are revised. You could fork the source to modify them to another value.  I do expose some server settings through a ServerLimits section of the configuration / builder and queue size can be overridden but not the min sampling intervals, so I may expose these too. Note I haven't tried with super fast polling rates so there might be latencies in there.

morphace, 2021-03-10:
Thanks a lot for the hint! I can see minimum sampling interval and publishing interval being adjustable in the configuration:
```

morphace, 2021-04-16:
Hi, any ideas?

locka99, 2021-04-20:
I've checked in changes to ensure the sanitize functions use the config settings

AiyionPrime, 2024-06-17:
Hey there :)
I'm glad you already thought about decreasing this limit.
Is there anything I can do to make it configurable?
As first step I'd like to be able to configure it in the ms range.