locka99/opcua #98: Accept faulty certificate

pulzzedavid, 2021-03-10:
I thought setting trust_server_certs would allow faulty certificates to be accepted, but looks like it just disregard keys?
Whether it is the IP address that is not matching the cert, or other details of the cert, I am wondering whether there is an option to accept the certificate in ClientBuilder.

locka99, 2021-03-10:
Check these first:

1. That the cert isn't already in the pki/rejected folder. e.g. maybe you set trust_server_certs to true but the pki had rejected before that flag was set. If the cert is already there then it will be untrusted even if that flag is later set. You should try deleting the value in the pki/rejected and try again. The second and subsequent times it should go straight into pki/trusted
2. That the cert can be loaded. If the cert cannot be parsed as an X509 certificate then it will fail before the point of deciding whether to trust it or not.

If it's not working even after these two things, set your logging to debug level and see if you can capture the code which includes "Validating cert with name on disk" to see what is tripping it up.

pulzzedavid, 2021-03-11:
Thanks for the information. I had checked those out but let me look into them again since it sounds like trust_server_certs should be doing the trick.

locka99, 2022-02-20:
Check if you have an update, otherwise I'll close