locka99/opcua #104: Spec compliance

NOP0, 2021-04-01:
Hi, more a question, than an issue. Is this library somehow compliant against the opc ua spec, is it "Certified" or similar? Im not sure whats normal for other opc ua implementations, only open62541 mentions that it's "certified"...thanks.

locka99, 2021-04-01:
No, the library is not certified but generally speaking it should be close to embedded level compliance. I think open62541 has run the OPCUA CTT (compliance testing tools) on itself and become certified. Unfortunately the CTT & certification isn't free which is a shame because it discourages projects that don't have the backing to afford it.

locka99, 2021-04-01:
I should add that OPCUA isn't just a thing which you pass or fail but at various levels of profile. So it's possible to be compliant with one profile but fall short of another. I think for most users the embedded profile is sufficient although there are things in the standard profile which are nice to have.

NOP0, 2021-04-02:
Thanks for your detailed answer! I agree its a shame that the certification tools are not free

closed 2021-04-02 by NOP0