locka99/opcua #106: Session activity keep-alive request not send on time

velichko, 2021-04-07:
I use the mainline and try to use the client.
1) Create and open Session from client side 
2) It is active and there is no activity from the client side for 60 seconds (no subscription, no reading, no writing)
3) After token_lifetime (60 000ms ) server closes the session

OPC UA Server gives SecureChannel has timed out

From opcua client side log:
2021-04-07 06:31:57.729 - INFO - opcua_client::comms::tcp_transport - ReadState has dropped
2021-04-07 06:31:57.730 - DEBUG - opcua_client::comms::tcp_transport - Read loop finished
2021-04-07 06:31:57.730 - DEBUG - opcua_core::runtime - deregistering component read-task, 2

locka99, 2021-04-19:
A workaround is to put a periodic call to read a known value value in the server just to keep the connection alive. It should probably do this internally.

locka99, 2021-04-19:
The keepalive would be to renew the secure channel before it expires, but the easiest way from the client code side is a benign read operation, e.g. to get a property from the root folder or server.

locka99, 2021-04-19:
I actually have a spawn_session_activity_task() that does a read on an interval for this purpose but maybe it isn't firing.

MatthiasEckhart, 2022-11-08:
Are there any updates on this issue?
I am currently running into the same problem: When the server closes the secure channel after the requested lifetime (60000 ms) as expected, the client is not able to renew the secure channel when issuing the next OPC UA request (after inactivity):

```
 TRACE opcua::client::session::session_state > issue_or_renew_secure_channel(Renew)
 INFO  opcua::client::session::session_state > Making secure channel request
 INFO  opcua::client::session::session_state > security_mode = None
 INFO  opcua::client::session::session_state > security_policy = None
 TRACE opcua::client::message_queue          > Sending request OpenSecureChannelRequest(OpenSecureChannelRequest { request_header: RequestHeader { authentication_token: NodeId { namespace: 1, identifier: Guid(13ae1e7c-10e3-57c3-98e9-eefe5bd5bdfb) }, timestamp: DateTime { date_time: 2022-11-08T14:08:45.807376900Z }, request_handle: 5, return_diagnostics: (empty), audit_entry_id: UAString { value: None }, timeout_hint: 10000, additional_header: ExtensionObject { node_id: NodeId { namespace: 0, identifier: Numeric(0) }, body: None } }, client_protocol_version: 0, request_type: Renew, security_mode: None, client_nonce: ByteString { value: None }, requested_lifetime: 60000 }) to be sent
 ERROR opcua::client::message_queue          > Send message will fail because sender has been closed
```

I am using version 0.11.