locka99/opcua #120: [Help needed]: How to borrow variable of Server type with threads to work with infinite loops?

jigar88, 2021-06-30:
Hello I am trying to create lot of nodes from the messages receiving from tcp server.  I tried to follow demo server code but I have an issue for borrowing the server variable. I am newbie to rust as well.  Can someone help to resolve the issue. I have continuous loop in the one of the rust file so opc server never gets start.  This is more related to rust but due to opc code involvement I just want to get help from here if someone can. 

Module: simulator
--------------------
main.rs
--------
```
mod client;

fn main() {
    unix_datagram_client::tcp_datagram_client()
}
``` 

client.rs
--------
```
use std::os::unix::net::UnixDatagram;
use std::{thread, time};
use serde_json::json;

pub fn tcp_datagram_client() {
    pub static FILE_PATH: &'static str = "/tmp/socket.sock";
    let socket = UnixDatagram::unbound().unwrap();
    match socket.connect(FILE_PATH) {
        Ok(socket) => socket,
        Err(e) => {
            println!("Couldn't connect: {:?}", e);
            return;
        }
    };
    println!("TCP client Connected to TCP Server {:?}", socket);
    loop {
        let message_payload = json!({
            "name": "NJ/Mercer/Pennington/street1",
            "value":"Isa",
            "status": "close",
        });
        socket.send(message_payload.to_string().as_bytes()).expect("recv function failed");
        thread::sleep(time::Duration::from_secs(1));
    }
}
```

Module: src
-------------
main.rs
--------
```
mod server;
mod process_tcp_messages;
mod opc_server;
mod global_nodes;

fn main() {
   opc_server::run_opc_server();
}
```

opc_server.rs
---------------
in this code is there a better way I can use server variable, if I try to create a thread it gives error of can't borrow moved variable. 
```
use opcua_server::prelude::*;
use std::path::PathBuf;
use std::thread;
use std::borrow::BorrowMut;

use crate::global_opc_nodes;
use crate::process_tcp_messages;

pub fn run_opc_server() {
    let mut server = Server::new(ServerConfig::load(&PathBuf::from("/tmp/server.conf")).unwrap());

    let ns = {
        let address_space = server.address_space();
        let mut address_space = address_space.write().unwrap();
        address_space.register_namespace("OPCUA-SERVER").unwrap()
    };

    global_nodes::create_global_variables(&mut server, ns);
    process_incoming_tcp_messages::run(&mut server, ns);

    server.run();
    println!("OPC Server stared and running...");

}
```

global_nodes.rs
-----------------
```
use opcua_server::prelude::*;
use std::path::PathBuf;
use std::thread;

static ROOT_NODE_PATH: &'static str = "OPC UA SAMPLE SERVER";

pub fn create_global_variables(server: &mut Server, ns: u16) {
    let address_space = server.address_space();
    let mut address_space = address_space.write().unwrap();

    let metrics_folder = address_space.add_folder("Folder",
                                                  "Folder",
                                                  &NodeId::objects_folder_id()).unwrap();
    let test_node = NodeId::new(ns, UAString::from("metrics"));
    let _ = address_space.add_variables(
        vec![
            Variable::new(&test_node, "metrics", "metrics", 0 as i32),
        ], &metrics_folder,
    );

}
```

process_tcp_messages.rs
----------------------------
I have  a  infinite loop here to process tcp server messages received from client. I try to put this code in thread but did not work , may be I am not doing proper threading.  Based on the received message, I will create nodes and update its values continuously.  Currently that function is comment out. 
```
use std::thread;
use std::sync::mpsc;
use serde_json::{Value, from_str};
use opcua_server::server::Server;
use std::ops::DerefMut;
use std::borrow::BorrowMut;

use crate::unix_datagram_server;

static ROOT_NODE_PATH: &'static str = "OPC UA SAMPLE SERVER";

pub fn run(server: &mut Server, ns: u16) {
    let (tx_to_opc_server, rx_from_opc_server) = mpsc::channel();
    thread::spawn(|| unix_datagram_server::tcp_datagram_server(tx_to_opc_server));
    println!("TCP server started...");
    loop {
        let message_from_tcp_server = rx_from_opc_server.recv().unwrap();
        let closing_bracket_index = message_from_tcp_server.
            as_bytes().
            iter().
            position(|&x| x == b'}').
            map(|i| i + 1).
            unwrap_or_else(|| message_from_tcp_server.len());
        let message_json: Value = serde_json::from_str(&message_from_tcp_server[..closing_bracket_index]).unwrap();
        let mut path = message_json["name"].clone().to_string();
        path = path.replace("\"", "");
        let tag_to_update = ROOT_NODE_PATH.to_owned() + &*path;
        let split_path: Vec<&str> = path.as_str().split("/").collect(); // this gives iterator, convert this to vector using collect()
        let tag_status = message_json["status"].clone();
        let tag_value = message_json["value"].clone();
        println!("{:?}, {:?}, {}, {}", tag_to_update, split_path, tag_status, tag_value);
        // create_opc_nodes::create_nodes_recursively(server.borrow_mut(), split_path, tag_value, ns);
    };
}
```

schroeder-, 2021-06-30:
Use _Server::run_server_ and Arc<RwLock<Server>>
quick example (not complete)

```
    let server = Arc::new(RwLock::new(new_server(0)));
    let server2 = server.clone();
    let (tx, rx) = channel();
    let _t = thread::spawn(move || {
        // This should run & block until it is told to abort
        Server::run_server(server);
        tx.send(()).unwrap();
    });

    {
        let server = server2.write().unwrap();
        ...
    }
```

jigar88, 2021-06-30:
@schroeder- I am not quite getting it, but I will keep tying to troubleshoot.

schroeder-, 2021-06-30:
If you need to store a variable on multiple times with different lifetimes you normally store the variable as refcounted variable. If you are single threaded you use Rc<T>, for use in multiple threads you use Arc<T>. This way you can work around your the lifetime issue. To secure share a variable you use a RwLock<T> or Mutex<T>. In this context you should use Arc<RwLock<Server>>, because the server needs this to use with Server::run_server.

locka99, 2021-07-01:
When you call server.run() it basically blocks the current thread so nothing else will happen until it exits. If you want to do other stuff then spawn a thread and run the server from that and carry on in the other thread.

It's hard to say what you want to do from the rest of the code, but you there are two ways of setting values on variables in the server - either you can just set the value and let the server manage it or you can register a getter which is called to retrieve the value that you manage.

The simple_server demonstrates both of these which I called push and pull there. It really depends what you're doing which is the best approach but I provide both as ways to do it.

jigar88, 2021-07-02:
Okay I am trying to use Arc with Mutex to reuse server variable multiple time.

```
pub fn run_opc_server() {
    let mut server = Server::new(ServerConfig::load(&PathBuf::from("/tmp/server.conf")).unwrap());
    let server2 = Arc::new(Mutex::new(server));
    let server2 = Arc::clone(&server2);
    let ns = {
        let address_space = server2.lock().unwrap().address_space();
        let mut address_space = address_space.write().unwrap();
        address_space.register_namespace("EMS").unwrap()
    };
    global_opc_nodes::create_global_variables(&mut server2.lock().unwrap(), ns);
    server2.lock().unwrap().run();
```

This gives an error 

```
error[E0507]: cannot move out of dereference of `std::sync::MutexGuard<'_, opcua_server::server::Server>`
  --> src/opc_server.rs:22:5
   |
22 |     server2.lock().unwrap().run();
   |     ^^^^^^^^^^^^^^^^^^^^^^^ move occurs because value has type `opcua_server::server::Server`, which does not implement the `Copy` trait
```

schroeder-, 2021-07-02:
You have to call Server::run_server, if you want to use an Arc<RwLock<Server>>. 
```
let server2 = Arc(RwLock::new(.........));
Server::run_server(server2);
```

jigar88, 2021-07-02:
Ok , I have reached halfway. Now opc server starts but then I want to use same server variable for tcp server loop. But now it gives error that variable moved to closure so you can't borrow.

````
pub fn run_opc_server() {
    let server = Server::new(ServerConfig::load(&PathBuf::from("/tmp/server.conf")).unwrap());
    let server2 = Arc::new(RwLock::new(server));
    let server2 = Arc::clone(&server2);
    let ns = {
        let address_space = server2.write().unwrap().address_space();
        let mut address_space = address_space.write().unwrap();
        address_space.register_namespace("EMS").unwrap()
    };
    global_opc_nodes::create_global_variables(&mut server2.write().unwrap(), ns);
    thread::spawn(move || Server::run_server(server2));
    println!("OPC Server stared and running...");
    process_incoming_tcp_messages::run(&mut server2.write().unwrap(), ns);
}

gives error:

```
error[E0382]: borrow of moved value: `server2`
  --> src/opc_server.rs:21:46
   |
12 |     let server2 = Arc::clone(&server2);
   |         ------- move occurs because `server2` has type `std::sync::Arc<std::sync::RwLock<opcua_server::server::Server>>`, which does not implement the `Copy` trait
...
19 |     thread::spawn(move || Server::run_server(server2));
   |                   -------                    ------- variable moved due to use in closure
   |                   |
   |                   value moved into closure here
20 |     println!("OPC Server stared and running...");
21 |     process_incoming_tcp_messages::run(&mut server2.write().unwrap(), ns);
   |                                              ^^^^^^^ value borrowed here after move
```
```

schroeder-, 2021-07-02:
Maybe you should read about how to use shared state in rust:
[rwlock](https://riptutorial.com/rust/example/24527/read-write-locks)
[rust book](https://doc.rust-lang.org/book/ch16-03-shared-state.html)

For your problem you can clone an Arc before spawn a thread.

```
pub fn run_opc_server() {
    let server = Server::new(ServerConfig::load(&PathBuf::from("/tmp/server.conf")).unwrap());
    let server2 = Arc::new(RwLock::new(server));
    let ns = {
        let address_space = server2.write().unwrap().address_space();
        let mut address_space = address_space.write().unwrap();
        address_space.register_namespace("EMS").unwrap()
    };
    global_opc_nodes::create_global_variables(&mut server2.write().unwrap(), ns);
   let server_th = server2.clone();
    thread::spawn(move || Server::run_server(server_th ));
    println!("OPC Server stared and running...");
    process_incoming_tcp_messages::run(&mut server2.write().unwrap(), ns);
}
```

jigar88, 2021-07-02:
so I have both servers working now, its just sometimes not connecting in time. opc server takes little longer to connect but I am able to process messages on opc server from tcp server. I ended up doing 

```
pub fn run_opc_server() {
    let opc_server = Server::new(ServerConfig::load(&PathBuf::from("/tmp/server.conf")).unwrap());
    let opc_server_arc_object = Arc::new(RwLock::new(opc_server));
    let ns = {
        let address_space = opc_server_arc_object.write().unwrap().address_space();
        let mut address_space = address_space.write().unwrap();
        address_space.register_namespace("EMS").unwrap()
    };
    global_opc_nodes::create_global_variables(&mut opc_server_arc_object.write().unwrap(), ns);
    let threaded_server_object = opc_server_arc_object.clone();
    thread::spawn(move || Server::run_server(threaded_server_object));
    // let sts = NodeId::new(0, 2256);
    // let v = DataValue::new_now(sts.clone());
    loop {
        thread::sleep(time::Duration::from_secs(30));
        let threaded_server_for_nodes = opc_server_arc_object.clone();
        thread::spawn(move || process_incoming_tcp_messages::run(&mut threaded_server_for_nodes.write().unwrap(), ns));
    }
}
```
Now I want to process message to structure the opc nodes. for example I am receiving message like 
```
"NJ/Mercer/Pennington/street1/isa"
```
then I am splitting the string in the vector like 
```
["NJ", "Mercer", "Pennington",  "street1", "isa"]
["NJ", "Mercer", "Pennington",  "street1", "emily"]
```
Now I am looping through vector elements and want to create nodes as folders for (vector length -1 )  in the structure like
```
NJ  - folder
  Mercer  - folder
	Pennington  - folder
		Street1 - folder
                      isa - Variable
                      emily -variable
```
so street folder can have multiple variables based on messages from tcp. so if folders are there then don't recreate but just add variable and if variable also in the folder then update its value.

Also I want to assign variant type for variable based on the variable value, for that I have a function already but not sure how can I set variable type.

I was able to achieve this functionality in python with node class. I tried to use address_space.find_hierarchical_references(&nodeid) from this library but its not finding any nodes , it just returns None. so I can't create a node structure I want.  Can you guys direct me ?

Here is the rust code.

```
pub fn get_data_type(tag_value: Value) -> VariantTypeId {
    match tag_value {
        Value::Null => VariantTypeId::Empty,
        Value::Bool(_b) => VariantTypeId::Boolean,
        Value::Number(_n) => VariantTypeId::Float,
        Value::String(_s) => VariantTypeId::String,
        _ => VariantTypeId::Empty,
    }
}

pub fn create_nodes_recursively(server: &mut Server, message_chunks: Vec<&str>, tag_value: Value, ns: u16) {
    let address_space = server.address_space();
    let mut address_space = address_space.write().unwrap();
    let set_variable_data_type = get_data_type(tag_value.clone());
    let mut i = 0;
    // let mut tag_path = ROOT_NODE_PATH.to_owned() + "/";
    address_space.find_hierarchical_references()
    let mut tag_path = "".to_string();
    while i < message_chunks.len(){
        tag_path += message_chunks[i];
        let make_node_id = NodeId::new(ns, UAString::from(tag_path.clone()));
        // println!("{:?}", make_node_id.clone());
        if i < message_chunks.len() -1 {
            address_space.add_folder(tag_path.clone(), tag_path.clone(), &NodeId::objects_folder_id()).unwrap();
            tag_path += "/";
        }
        i += 1
    }
}
```

Here is the working python code with node class.

```
def get_tag_data_type(tag_value):
    if type(tag_value) is bool:
        return VariantType.Boolean
    elif type(tag_value) is int:
        return VariantType.Int64
    elif type(tag_value) is float:
        return VariantType.Float
    else:
        return VariantType.String

  def create_nodes_recursively(self, messageChunks, tagValue):
      rootNode = self.root_node
      tagPath = self.root_node_path
      set_data_type = get_tag_data_type(tagValue)
      i = 0
      while i < len(messageChunks):
          try:
              tagPath += messageChunks[i]
              node = self.server.get_node(rootNode)
              if Node(self.server.iserver.isession, tagPath) not in node.get_children():
                  if i != len(messageChunks) - 1:
                      node.add_folder(tagPath, messageChunks[i])
                  else:
                      node.add_variable(tagPath, messageChunks[-1], tagValue, set_data_type)
                      Node(self.server.iserver.isession, self.queueSizePathNodesCreated).set_value(self.index)
                      if self.writeToDynamo:
                          self.db_put.process_dynamodb(tagPath, str(tagValue))
                      self.index += 1
          except uaerrors.BadNodeIdExists:
              pass
          except uaerrors.BadParentNodeIdInvalid:
              pass
          rootNode = tagPath
          tagPath += "/"
          i += 1

```

locka99, 2021-07-10:
Closing because remaining issues are probably race conditions in the implementation using the crate

closed 2021-07-10 by locka99