locka99/opcua #122: OPC-UA PubSub Support?

sysarcher, 2021-07-05:
Does this project support OPC UA PubSub as described here: https://reference.opcfoundation.org/v104/Core/docs/Part14/5.1/

schroeder-, 2021-07-05:
Pubsub is not implemented atm.
I started to work on an implementation, but it is work in progress and the api surface will likely change. [opcua_pubsub](https://github.com/schroeder-/opcua_pubsub)

sysarcher, 2021-07-05:
AWESOME @schroeder- .... 

I'll try your stuff out if I can (right now, I'm too new to this field)... but, is the goal to submit a PR to this here repo in the end??

schroeder-, 2021-07-05:
In the long run, for sure. But there is a lot of work left before that. The hole information model is missing and I need a good solution to schedule all tasks.

closed 2021-07-06 by sysarcher