locka99/opcua #124: [Help] Deadlock while writing to variable?

jigar88, 2021-07-12:
I have server side running with some variables. I am trying to update the variable value but it results in deadlock in thread. How can I update the variable value with RwLockWriteGuard presence ?

I have set my server and address space as shown below.
```
let opc_server = Server::new(ServerConfig::load(&PathBuf::from("/tmp/server.conf")).unwrap());
let server = Arc::new(RwLock::new(opc_server));
let address_space = server.address_space();
let mut address_space = address_space.write().unwrap();
let now = DateTime::now();
address_space.set_variable_value(&NodeId, valueToUpdate, &now, &now);
```

this results in 
```
thread '<unnamed>' panicked at 'rwlock write lock would result in deadlock', /build/rustc-UHkz09/rustc-1.47.0+dfsg1+llvm/library/std/src/sys/unix/rwlock.rs:86:13
```

jigar88, 2021-07-12:
this might be causing by other piece of code. Not related to opcua code.

closed 2021-07-12 by jigar88

jigar88, 2021-07-13:
```
let server = Arc::new(RwLock::new(opc_server));
let address_space = {
   let server = server.read().unwrap();
   server.address_space()
};

```
This will return address_space as function so can I reuse it as an address space to pass an parameter for later function or I need to recreate it ?

locka99, 2021-07-13:
That block unlocks the server, calls the function address_space(), implicitly releases the lock, and returns a Arc<RwLock<AddressSpace>> to the outer let statement where it is assigned. 

I would try the simple-server too and see if it suffers the issue you are seeing since it also uses set_variable_value()

jigar88, 2021-07-13:
is there a way to get server status in the tag for monitoring, like weather server is started or running or broken? Or I need to use server status node defined in the server state to get the server status ? In python there is a function like server.is_running() , is there anything equivalent available ?