locka99/opcua #125: Reading Multiple Node with session.read()

forcmti, 2021-07-12:
To read a node, u have to give ReadValueId to read() function and u will get Result with Data.

I am trying to read multiple node at the same time, i can write a loop and read the single node and result is obtained.
I was doing this method, but time taking is more than 2 seconds, so i want to try to read faster.

When i read the documentation, the read() function, it says input as 
`"nodes_to_read - A list of ReadValueId to be read by the server"`

So can we give list of ReadValueId or Single Value id?. Because when i try making list of ReadValueId with vector , read function is showing Error.

So is it possible to do it one read, or am i misunderstood the documentation.

The code i tried is given below

```
let nodes = ["/Channel/State/actFeedRateIpo","/Bag/State/opMode", "/Channel/MachineAxis/actFeedRate[u1, 1]"];
let readid:Vec<ReadValueId> = nodes.iter().map(|node| ReadValueId::from(NodeId::new(namespace, *node).into())).collect();
let read_val = session.read(readid);
```

Problem is, i dont know problem is with my code, or is it not possible to read that way.

schroeder-, 2021-07-12:
Reading a list of values, with read, is no problem. Whats the error, returned from read?

locka99, 2021-07-12:
The client's Session::read() takes a slice (array) of ReadValueIds and returns Result<Vec<DataValue>, StatusCode>. i.e if you pass it 3 things to read it should give you 3 results to process.

It should work fine unless you exceed the number of values that the server will process in a single read call in which case you will get an error.

forcmti, 2021-07-13:
Thanks for the input, My Problem was i was passing the complete array instead of slice, 
Changing into the slice of array, its working and time taken to read reduced from 2sec to 300ms

Also i didnt have to convert NodId to ReadValueId, it automatically does that.
So i changed the code to this and its working

```
let readid:Vec<ReadValueId> = nodes.iter().map(|node| NodeId::new(namespace, *node).into()).collect();
let read_val = session.read(&readid);
```

Thanks again for the help

locka99, 2021-07-13:
Glad to help!

closed 2021-07-13 by locka99