locka99/opcua #126: Getting Value from LocalizedText Variant

forcmti, 2021-07-13:
Some of the types variable getting from event Handler is of "LocalizedText" Variant.

To Obtain the actual message from the "LocalizedText", is there any method is there.
Only thing i can think of is converting into string and doing string manipulation to get the message.

Is there any proper method to get "text" property from LocalizedText.

For Example My Response from OPCUA Server for Message that i printed i got is

```
Message: LocalizedText(LocalizedText { locale: UAString { value: None }, text: UAString { value: Some("The alarm severity has increased.") } })
```

Is there any simple method i am missing, so i can get the "text" value from LocalizedText.
For UAString , i know by calling value function, you can get the String.

So for LocalizedText is there any method like that?

schroeder-, 2021-07-13:
You can directly access the `text` member of LocalizedText struct.
```
let text = LocalizedText::new(....);
let s = s.text.value();
```

schroeder-, 2021-07-13:
You can match a Variant for example: 
 ```
        match field{
            Variant::LocalizedText(text) => text.text.value(),
            Variant::String(text) => text.value(),
            _ => ...
        }
```
or use if let:

```
        if let Variant::LocalizedText(text) = field{
            ... 
        } else {
            ...
        }
```

forcmti, 2021-07-13:
Thanks man, Why didn't i think of this, now i have to think like this.
It worked perfectly.

closed 2021-07-13 by forcmti