locka99/opcua #127: [Help Required] How to get tag values related to server status?

jigar88, 2021-07-15:
is there a way to get server status in the tag for monitoring, like weather server is started or running or broken? Or I need to use server status node defined in the server state to get the server status ? In python there is a function like server.is_running() , is there anything equivalent available ?

also I tried to use server info in the Ignition client but most of the nodes are null and not showing data.
[image]

schroeder-, 2021-07-16:
If you are on the server side:
Get [server_state](https://docs.rs/opcua-server/0.8.0/opcua_server/server/struct.Server.html#method.server_state) and then use [is_running](https://docs.rs/opcua-server/0.8.0/opcua_server/state/struct.ServerState.html#method.is_running)

locka99, 2021-07-16:
I think if you're writing a server it is better to push state into OPC UA rather than try fetch state from it.

jigar88, 2021-07-16:
How do you push state ?

closed 2021-07-27 by jigar88