locka99/opcua #133: OPC UA client fails to store server certificate if common name (CN) is not empty on Linux

jonasar, 2021-08-12:
In function cert_file_name() in crypto/src/certificate_store.rs, the file name is concatenated from the certificate common name and the thumbprint, with spaces between them (!)·

The bug is on this line:
https://github.com/locka99/opcua/blob/1dbca0e24759355a77c1388f3f6d2883cd745127/crypto/src/certificate_store.rs#L424

Spaces in filenames are not valid on Linux.

This was tested on opcua_client v 0.8

Also if the common name includes invalid characters for filenames, e.g. forward slash "/", the certificate will not be stored correctly.

jonasar, 2021-08-12:
I can create a pull-request with a fix if you give me necessary permission on the repo.

I suggest replacing the spaces with dots and also skipping the square brackets "[" "]" in the file name.

locka99, 2021-08-12:
Most Linux filesystems supports pretty much every character with perhaps the exception of forward slashes or special names like "." and "..". So I'd accept a patch that strips out the "/" but other chars including whitespaces should remain the way they are. Whitespaces in Linux can be used although from a command prompt you might have to escape them, e.g. " " becomes "\ " as you type it.

locka99, 2021-08-12:
A further note - I don't know what the encoding rules are for X509 regarding unicode chars in the CN but Linux typically allows UTF-8 encoded filenames  so providing the field comes out as valid unicode we should also assume that to be a valid filename also aside from the forward slash.

jonasar, 2021-08-19:
You are right, whitespace is allowed in filenames and it was actually only the forward slash in the CN that caused the opcua client to fail when trying to store the server certificate.

jonasar, 2021-08-23:
Here is the patch:

```
diff --git a/crypto/src/certificate_store.rs b/crypto/src/certificate_store.rs
index ad757f62..8bd3cb2b 100644
--- a/crypto/src/certificate_store.rs
+++ b/crypto/src/certificate_store.rs
@@ -414,7 +414,7 @@ impl CertificateStore {
     /// the cert's common name being empty or not
     pub fn cert_file_name(cert: &X509) -> String {
         let prefix = if let Ok(common_name) = cert.common_name() {
-            common_name.trim().to_string()
+            common_name.trim().to_string().replace("/", "")
         } else {
             String::new()
         };

```

I would be very grateful if you would be willing to release a patch version based on 0.8 on crates.io including this fix soon.

locka99, 2021-08-24:
I'll publish 0.8.1 for this and it has landed on 0.9

closed 2021-08-24 by locka99