locka99/opcua #134: error: could not compile `opcua-types`

cradle1001, 2021-08-22:
error: could not compile `opcua-types`

Caused by:
  process didn't exit successfully: `rustc --crate-name opcua_types --edition=2018 types/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C embed-bitcode=no -C debuginfo=2 -C metadata=399a566fffb005e9 -C extra-filename=-399a566fffb005e9 --out-dir /var/www/opcua/target/debug/deps -C incremental=/var/www/opcua/target/debug/incremental -L dependency=/var/www/opcua/target/debug/deps --extern base64=/var/www/opcua/target/debug/deps/libbase64-e2a906ac29617111.rmeta --extern bitflags=/var/www/opcua/target/debug/deps/libbitflags-f89ae4b715176e2d.rmeta --extern byteorder=/var/www/opcua/target/debug/deps/libbyteorder-3a640e6d7cbe5c36.rmeta --extern chrono=/var/www/opcua/target/debug/deps/libchrono-f01329d9a0a1e8e8.rmeta --extern lazy_static=/var/www/opcua/target/debug/deps/liblazy_static-5c53d641eb010a81.rmeta --extern log=/var/www/opcua/target/debug/deps/liblog-413348ce8ba1e534.rmeta --extern regex=/var/www/opcua/target/debug/deps/libregex-786fb7048e33eaa9.rmeta --extern serde=/var/www/opcua/target/debug/deps/libserde-332a662ee12a03b3.rmeta --extern serde_derive=/var/www/opcua/target/debug/deps/libserde_derive-6f3ae2b6b2341058.so --extern uuid=/var/www/opcua/target/debug/deps/libuuid-bb898c12cd2dbe78.rmeta` (signal: 9, SIGKILL: kill)

schroeder-, 2021-08-22:
Must likely you are running out of memory. The think opcua-types needs about 2gb ram on my maschine.

locka99, 2021-08-22:
Please confirm if memory is the issue. I don't know the reasons rustc might need so much memory other than there are a lot of public types in that crate and Rust might hold them in all memory. Perhaps there are some optimizations that could bring memory consumption down a bit, e.g. using specific types in places instead of *.

cradle1001, 2021-08-23:
how can i compile in windows and run at Linux?

locka99, 2021-08-24:
I don't know if Rust will cross-compile to Linux, but you can install Windows Services for Linux (WSL2) on Windows 10 and run Ubuntu / Fedora straight from inside of it. So you can always do it that way.

milgner, 2021-11-26:
Since Rust creates self-contained binaries, it should be possible to cross-compile applications using this crate just like any other Rust project. You just need to add the target to your installation before you can use it. See https://rust-lang.github.io/rustup/cross-compilation.html for more information. In any case, there shouldn't be anything special in this crate that prevents cross-compilation.