locka99/opcua #135: [Help]: How to create server from ServerBuilder ?

jigar88, 2021-08-30:
Hello ,

I am trying to create an OPC server from ServerBuilder. I have copied details from the server.config file from the demo server but not able to create a server. Is there something I am missing. Here is the sample code for server creation .

```
let opc_server = ServerBuilder::new()
            .application_name("testOpcServer")
            .application_uri("urn:testopcServer")
            .discovery_urls(vec![endpoint_path.to_string()])
            .create_sample_keypair(true)
            .pki_dir("/tmp/pki")
            .user_token(sample_user_id, ServerUserToken::user_pass("user1", "user1pass"))
            .endpoints(
                [
                    ("basic256sha256_sign_encrypt", endpoint_path, SecurityPolicy::Basic256Sha256, MessageSecurityMode::SignAndEncrypt, &user_token_ids),
                ].iter().map(|v| {
                    (v.0.to_string(), ServerEndpoint::from((v.1, v.2, v.3, &v.4[..])))
                }).collect())
            .server().unwrap();
```

kimim, 2021-11-24:
Hello, since you are creating the server from `ServerBuilder`, you do not need configuration files.
The problem in the code is that `sample_user_id` is not defined. You could replace this to a string:

`            .user_token("sample_user_id", ServerUserToken::user_pass("user1", "user1pass"))`

locka99, 2021-11-24:
Probably the best way to start is to look at new_sample() in builder.rs since that produces a working sample server config and demonstrates how to programmatically create one. You can also prime the ServerBuilder by calling from_config(), supplying a ServerConfig you loaded from a path, or some hybrid. Just depends on which settings you want to program in and which you wish to load in.