locka99/opcua #137: Support intermediate chunks

milgner, 2021-09-23:
When trying to connect to a TCP-based UA server, depending on the vendor implementation, the error "We don't support intermediate chunks yet" is thrown.

When connecting to a NodeJS-based server, the connection could be established successfully, but when connecting to the [Prosys simulation server](https://opcfoundation.org/products/view/prosys-opc-ua-simulation-server), it fails with the above error.

What would need to be done to support intermediate chunks? I assume the previous chunks would need to be saved before processing the message, with only the final message triggering message processing - but there's probably more to that? I.e. detecting if follow-up chunks arrive or something like that?

schroeder-, 2021-09-23:
Which version are you using? Itermediate chunks are support with commit: https://github.com/locka99/opcua/commit/73d77d117b902f19e3e069624061393c1d8489e9

milgner, 2021-09-24:
For some reason I just ran into the same issue again, this time with version 0.8.1:

```
'We don't support intermediate chunks yet', /home/marcus/.cargo/registry/src/github.com-1ecc6299db9ec823/opcua-client-0.8.1/src/comms/tcp_transport.rs:83:17
```

It happens when calling `get_server_endpoints_from_url`.

locka99, 2021-09-26:
They are supported in 0.9.0 (the master branch) so I will close the issue as they will release when it does (no ETA)

closed 2021-09-26 by locka99