locka99/opcua #142: Issue with decoding limits (max string length) (Client)

webhet, 2021-11-03:
The OPCUA server which I connect to with the client library advertises a larger than normal MaxStringLength in it's ServerCapabilities.
When I try to call a method of which the result string is longer than 65535 I receive the following error message: 

```
opcua_types::string - String buf length 196181 exceeds decoding limit 65535
```

The MaxStringLength of the server is larger than that (16776192).
Is 65535 a desired upper limit for the client library or is there some way that the client enforces the server limits? 
If not is there any ability to configure the limit on the client side?

locka99, 2021-11-05:
At present decoding uses the MAX_STRING_LENGTH constant in opcua-types/src/lib.rs that you can change if you fork the source and modify it. This is set in a DecodingOptions which is passed into the (de)serialization functions which enforce the maximum lengths of things.

It would be nice to allow this value to be set via configuration and there is a TODO in the code where this needs to happen but it hasn't been done yet. So I'll leave the bug open until it is done. FYI this is the TODO

https://github.com/locka99/opcua/blob/master/client/src/session/session.rs#L179

Basically I need to create a config decoding_options block and options for the builder and set the values from this TODO instead of the DecodingOptions::default() that is initialised from constants. There may also be some low level cases, where for whatever reason the DecodingOptions::default() constructor must be used and I will have to appraise their impact.

milgner, 2021-11-26:
I encountered a similar situation and would be willing to assist with implementation. Let me know if there's anything I can do to help.