locka99/opcua #145: Why is Variant array_length == 0 a problem?

milgner, 2021-11-26:
Although I personally would prefer using `Variant::Empty` in case of an empty array, I yesterday encountered [this error](https://github.com/locka99/opcua/blob/4527d410782672fa731e526f922970448f084a93/types/src/variant.rs#L726) when our client tried to decode the response from a server that sent a zero-length array.

Now this code does look very explicit about its intent and I was wondering whether there's anything in the OPC UA spec that forbids arrays of length 0?

In https://open62541.org/doc/0.3/types.html#variant they seem to have many different variations on dealing with array length 0.

locka99, 2021-11-27:
Mostly it's because I can't tell you a type of an array if there is nothing in it to infer a type. I'm open to fixes that comply with the spec on this.

schroeder-, 2021-11-27:
Empty arrays are allowed. For decoding we can handle length 0 == null array (length=-1).

milgner, 2021-11-29:
Thank you for elaborating! I'll have a closer look at the source then and think about potential solutions.

schroeder-, 2021-11-29:
Found this in the specs for empty array, null array and Variant null: https://reference.opcfoundation.org/v105/Core/docs/Part6/5.1.9/
So null array == empty array is legal.

milgner, 2021-11-29:
Nice! That sounds like it would be also be in accordance with the specification if decoding would return `Variant::Empty` in case the array length is 0? :thinking:

schroeder-, 2021-11-29:
Its up to the implementation, if we want empty, null array and Variant::Empty to be decoded separate. But at least a comparison of Variant::Empty and empty/null array should return true.

PR #151 (2022-11-05): Refactor: zero length variant array decoding
Closes #145 

As I was just on the train with a few hours to kill, I took another look at this issue.

In the discussion, @locka99 pointed out that there were problems inferring the type of 0-length arrays. Yet after reviewing  https://reference.opcfoundation.org/v104/Core/docs/Part6/5.2.2/ it seems to me that, if the array bits are set, the type id should be there as well, even if the array length is 0. (the code being executed is in a branch where the array bit is set, so the encoding mask as a whole is not 0)
So I think the most likely explanatio [... cut]
closed 2022-11-05 by locka99