locka99/opcua #147: check if client connection is successfull

schroeder-, 2021-12-10:
Currently it is possible that Client **connect_to_endpoint** and **connect_to_endpoint_id** can fail and still return a session (See https://github.com/locka99/opcua/blob/89ea484783038f96448822c095c2b3acf4bb6719/client/src/client.rs#L221). You get the error only when you try to use any service. I think i would be better to return every error from connect_to_endpoint and connect_to_endpoint_id. Or at least give a possibility to check if the session is successful established, with out using a service. This would simplify the client logic.

locka99, 2021-12-20:
That line should probably return an Err(). I'll keep bug open until it's fixed

locka99, 2021-12-20:
Fixed, function will return an error status code if the connection failed, not a session

closed 2021-12-20 by locka99