locka99/opcua #150: UaException: Bad_NonceInvalid when connecting from inductiveautomation Ignition OPC Client

fautore, 2021-12-21:
Hi everybody,
I set up a opcua-demo-server as in `samples/demo-server` and used the default configuration, then i set up an Ignition server on the same machine and created a new OPC Client connection to localhost port 4855.

the endpoint i'm setting into ignition is the following:
`opc.tcp://localhost:4855/`
but the Ignition OPC Client keeps faulting with the following error message:
` UaException: status=Bad_NonceInvalid, message=nonce must be non-zero`

I also tried fiddling with the configuration of the demo-server, i tried dropping the endpoints that use encryption but i seem to have no luck with this.

I also set up a opcua-simple-client that i found on `samples/simple-client/`, default configuration and that seems to work fine.

Any ideas / help with the configuration i am running?
Any help will be appreciated,
Thank you.

schroeder-, 2021-12-21:
Which version are you using? Can you try current master? See issue #58 , which was fixed on master lately.

fautore, 2021-12-21:
Yes, downloaded master and compiled it yesterday.

schroeder-, 2021-12-21:
Maybe we have the same issue on the server side.
If you are using the configuration file try using this endpoint to force a password security policy:
```
 none:
    path: /
    security_policy: None
    security_mode: None
    security_level: 0
    password_security_policy: http://opcfoundation.org/UA/SecurityPolicy#Basic256
    user_token_ids:
      - ANONYMOUS
      - sample_password_user
      - sample_x509_user
```
Also does anonymous access with secruity_policy: None works?

fautore, 2021-12-21:
I tried just now with the configuration you suggested, same result, Ignition client side connection faults.

Also tried connecting from simple-client in localhost with "none" endpoint and it seems to work just fine.

milgner, 2022-09-13:
Not sure whether this is still current, but I encountered similar issues with the Prosys Simulation Server. The server was confused because the client sent a security certificate but no nonce. So instead of ignoring it for SecurityPolicy None, it complained.

There's a patch at https://github.com/re-gmbh/opcua/commit/2f6c80b9d0a1b7435f9b8d2661c5edf658e19cf9, I'll try to allocate a few minutes to create PR for it.