locka99/opcua #158: Does this library support HistoryRead

liuweiXX, 2022-01-18:
I tried toUaExpert to read a historical data using UaExpert, but fails
[image]

schroeder-, 2022-01-18:
Out of the box there is no support for HistoryReadRaw. But you can hook your own history backend into the server see:
[historical.rs](https://github.com/locka99/opcua/blob/64844b9e6de7c2dbd8aacfa48a4c2d57aa3e96ac/samples/demo-server/src/historical.rs)
There is just a barebone and returns not supported. But you can write your own logic to provide history values.

locka99, 2022-01-19:
Yes it's something you need to implement. The server could conceivably offer some kind of simple history backend but at the moment leaves it up to the implementation to do something with the call.

locka99, 2022-02-20:
Closing since the answer is yes

closed 2022-02-20 by locka99