locka99/opcua #160: CloseSessionRequest when not necesary?

oroulet, 2022-02-04:
Wrote my first rust ua client today. something simple: But the log show there is an extra ClsoeSessionRequest BEFORE the session is created and the client get a ServiceFault error. Do I do something wrong is that a bug?

```

        // Make the client configuration
        let mut client = ClientBuilder::new()
            .application_name("Simple Client")
            .application_uri("urn:SimpleClient")
            .product_uri("urn:SimpleClient")
            .trust_server_certs(true)
            .create_sample_keypair(true)
            .session_retry_limit(3)
            .client()
            .unwrap();

        if let Ok(session) = client.connect_to_endpoint(
            (
                args.url.as_ref(),
                SecurityPolicy::None.to_str(),
                MessageSecurityMode::None,
                UserTokenPolicy::anonymous(),
            ),
            IdentityToken::Anonymous,
        ) {
            let session_tx = Session::run_async(session.clone());
            let nid = NodeId::new::<u32>(2, 13);
            loop {
                let res = session.read().unwrap().read(
                    &[ReadValueId::from(&nid)],
                    TimestampsToReturn::Neither,
                    9999.0,
                );
                dbg!(&res);
                std::thread::sleep(std::time::Duration::from_millis(1000));
                //let _ = Session::run(session);
            }
            //session_tx.send(SessionCommand.stop());
        }
```

[image]

milgner, 2022-02-28:
I adressed this in https://github.com/locka99/opcua/pull/156

It happens because the endpoint enumeration uses a dummy session (session id 0) to communicate with the server. Maybe this could be further decoupled to avoid having sessions when there's no actual session required but OTOH this is a bit of an edge case.

locka99, 2022-03-03:
I've merged that PR to fix this. I see it was breaking appveyor so we'll see if that persists in which case a test may have to be fixed.

closed 2022-03-03 by locka99